pinout-xyz/Pinout.xyz #179: Add Energenie Pi-mote boards?

sheffieldnick, 2017-01-24:
Could these power remote control/monitoring boards be added? [https://energenie4u.co.uk/catalogue/category/Raspberry-Pi-Accessories](url)

I think they do:

1. RF transmitter only, [https://energenie4u.co.uk/res/pdfs/ENER314%20UM.pdf](url)
2. IR transmit/receive, [https://energenie4u.co.uk/res/pdfs/ENER314-IR_User_guide_V3.pdf](url)
3. RF transmit/receive, [https://energenie4u.co.uk/res/pdfs/ENER314-RT%20User%20Guide%20V2.pdf](url)

Many thanks

Gadgetoid, 2017-02-15:
I see absolutely no reason why not! Would you like to take a crash course in GitHub and try to submit them yourself? That way your contributions would go down on permanent record, and you'll have learned a useful skill/paved the way to future open-source contributions!

You don't *have* to, but the new code review features make it easy for me to guide and support contributors.

sheffieldnick, 2017-02-15:
Go on then. Got a link to a Github tutorial?

RogueM, 2017-04-01:
the Pi-Mote is now documented. If you'd like to submit other Energenie boards, the best is probably to git clone this repo, then duplicate [rhis file](https://raw.githubusercontent.com/Gadgetoid/Pinout.xyz/master/src/en/overlay/energenie-pi-mote.md), edit what is required in place and finally submit a pull request.

closed 2018-01-25 by Gadgetoid