pinout-xyz/Pinout.xyz #181: SenseHat pinout

framboise314, 2017-01-25:
Le brochage de la SenseHAT indique la connexion à I2C 
mais le site [ModMyPi](https://www.modmypi.com/raspberry-pi/breakout-boards/raspberry-pi-(official)/raspberry-pi-sense-hat) indique que SPI est connecté pour permettre la programmation de l'ATMEL AT TIny
Le [schéma de la Sense HAT](https://www.raspberrypi.org/wp-content/uploads/2015/08/Sense-HAT-V1_0.pdf) montre que le bus SPI est utilisé : 
Pins 19 21 23 22 et 24
les INT sont sur les broches 16 et 18
un de mes lecteurs m'a contacté à cause d'un conflit entre Piface Digital 2 (bus SPI) et SenseHAT...
ceci peut expliquer cela :)
cordialement
François
framboise314.fr

RogueM, 2017-01-25:
l'Atmel peut-etre reprogrammé par le biais du bus SPI, mais je ne pense pas que cela soit un probleme en soi.

... je n'ai pas de Piface Digital pour tester mais piloter un périphérique SPI avec 'chip select' sur bcm8 (comme le Sense-HAT) tel que le Display-O-Tron HAT est tout a fait possible, ce qui veut dire que l'EEPROM du Sense HAT ne bloque pas le bus.

Par contre je peux confirmer une issue avec un Blinkt, qui utlisise BCM23-24 (phys 16+18), donc les broches interrupts certainement ont l'air de créer une interférence.

Je vais essayer de trouver des détails supplémentaires sur le Piface et voir quelles broches il utlise... au pire cependant, il devrait etre possible de connecter son CS sur BCM7 (CS1) pour contourner le conflict.

RogueM, 2017-01-25:
ah, je vois! malheureusement le probleme est inverse, piloter un périphérique SPI peut bloquer la matrice du Sense-HAT.

... c'est vraiment bien dommage que la fondation n'a pas daigner eviter ce probleme en utilisant BCM7 pour la reprogrammation (ce qui n'aurait abslument rien changer pour eux!).

OK, je vais voir comment documenter ceci au mieux - la vocation de pinout étant de servir aussi bien les usagers avancés que débutants je ne suis pas sur de comment approcher le probleme, qui est facile a contourner et il serait dommage que les débutants considere la combinaison que votre lecteur envisagait impossible.

framboise314, 2017-01-25:
merci :)
bonne soirée !

closed 2017-01-27 by RogueM