pinout-xyz/Pinout.xyz #185: Unicorn Hats require Ground at pin 9

Pyroseza, 2017-02-13:
Title says it all

RogueM, 2017-02-14:
is this aquestion or an affirmation?

... but, yes, affirmative:
http://pinout.xyz/pinout/unicorn_hat

Pyroseza, 2017-02-14:
An affirmation because I am trying to put jumpers between the hat and the pi so that I can use a motor hat alongside and couldn't figure out what was happening until I plugged the ground wire. Also read that all grounds are linked so it does not have to be at pin 9 bit can be any ground.

RogueM, 2017-02-14:
hum, I'm pretty sure pin9 is a requirement. Some other may be connected but I don't think any other can be connected by themselves and the setup to work.

Pyroseza, 2017-02-14:
Interesting, I'll test it and let you know. Do you think it's worth
mentioning on the pages as requirement? I think it would have been quit
handy to see it at a glance instead of googling for an hour.

RogueM, 2017-02-14:
that is what the blob means... obviously all and any elctronic board requires vcc and ground. the pinout tell you all this info already, including which ground are electronically connected.

Pyroseza, 2017-02-14:
Interesting I'll go read it now, I must've missed it. 

Can you recommend a decent place on the web to read about gpio pins on the raspi? 

I think I just confused myself earlier, also this is my first attempt with electronics. So I'm quite noob and have already burnt out 3 or so components for putting them in the wrong way.

Kind Regards,
Jarrod Price

--------

RogueM, 2017-02-14:
I would recommend http://raspi.tv/rpi-gpio

closed 2017-03-16 by RogueM