pinout-xyz/Pinout.xyz #186: Please add the motor hat for the camjam edukit 3

Pyroseza, 2017-02-14:
Link to page https://camjam.me/?page_id=1035

I can't find the details on what it requires I hope you can, that would help me a lot.

RogueM, 2017-02-14:
this info can be found in [Worksheet: Running the Motors PDF](https://github.com/CamJam-EduKit/EduKit3/raw/master/CamJam%20EduKit%203%20-%20Robotics%20Worksheet%203%20-%20Running%20the%20Motors.pdf)

Pyroseza, 2017-02-14:
I've looked at this over and over and can't get it to work with those.

I'm trying different combinations of the power pins and ground as well but
am not coming right :(

I've also asked 4tronix for the schematics but nothing as yet from them.

Pyroseza, 2017-02-14:
Finally figured it!!!

Okay so the doc only says pins BCM 7 (pin 26), BCM 8 (pin 24), BCM 9 (pin
21) & BCM 10 (pin 19).

What they don't say is you can use any ground but most importantly the 3v3
power at pin 17!

Damn! Okay, I'm all good! Onwards and forwards full steam ahead, that's 2
hats so far now I got some sensors to configure now, no more hats phew....

Anyways thanks for the assistance will definitely go read up a lot more in
the GPIOs! XD

RogueM, 2017-02-14:
note that 'HAT' is a specific type of add-ons. The Unicorn is a HAT, the Cam Jam motor controller isn't... pedantic perhaps, but worth noting! ;-)

Pyroseza, 2017-02-14:
Interesting... okay to me a hat is just something that you stick on top over the GPIO pins. A bit ignorant of me making assumptions like that. I've only just got into all of this.

I appreciate the correction, thank you. 

Any other advice is welcome.

Kind Regards,
Jarrod Price

--------

RogueM, 2017-02-14:
it's not a hugely important distinction when you start, but it's good to know from the start that there are technical specifications that define a HAT.

Specifically, HATs attached to the top of the GPIO header and are not meant to be combined with other things, while also expected to support some level of 'autoconfiguration' (though very few do).

... of course you'll find very few pinout.xzy users, let alone @Gadgetoid and myself, to tell you off for not playing by the rule of 'one Pi, one HAT. It's in fact the very reason for pinout to exist!

if this subject piked your imterest, here's a useful link that tries to outline the philosophy for HATs:
https://www.raspberrypi.org/blog/introducing-raspberry-pi-hats/

Pyroseza, 2017-02-14:
Thanks for the feedback.

I really like the idea of combining components to create hats however I don't see why I should only have 1 i.e. block access to prevent me from using the other unused pins.

I feel like if I have 40 pins then I'm gonna use them all not just 3 (unicorn phat).

But anyways at least I got it working the way I wanted, so I'll have a look and see if I can desolder the headers later so that it is just the boards then I can squeeze them nice and snug into my robot.

I'm very happy to see a site like pinout.xyz, I'm looking for mor places to gather good and current up to date information especially for noobs like me with no prior experience in electronics.

A lot of it seems to be outdated for the older pi's, very specific to some other setup that does not pertain to me or is very complicated where I think it should be a lot simpler.

Like I dunno if it is just me... but are people over complicating things unnecessarily???

I feel like we should be making things easier and simpler but anyways... I can keep rambling on. 

Thanks for the advice, I've learnt more about the raspi today than I have in the last few months.

Cheers

Kind Regards,
Jarrod Price

--------

closed 2017-02-15 by RogueM

lurch, 2020-05-04:
Added in #362 :slightly_smiling_face: