pinout-xyz/Pinout.xyz #187: HTTPS should be default

diegorodriguezv, 2017-02-15:
Or another way to put it, HTTP pages should redirect to it's HTTPS counterpart.

HTTPS is already working (I checked https://pinout.xyz).

Browsers are getting constantly taking measures to differentiate secure from not secure sites. For example when I visit http://pinout.xyz/ in Chrome, the icon to the left of the address tells me that this site is not secure. This can discourage potential visitors.

I thought about putting this suggestion in the 2017 project but it seems to me that fixing this requires very little effort and can establish significant rapport with the security aware nerds.

Also. Remember Snowden...

Gadgetoid, 2017-02-15:
I think our use of CloudFlare's caching proxy makes this difficult- the vanilla server currently isn't configured to use SSH and I think a 301 redirect would only affect the requests from CloudFlare -> Our Server and not from the web. I've reached out to CloudFlare for help, maybe there's a solution!

Gadgetoid, 2017-02-15:
Apparently there is, and it's simple: https://twitter.com/matthewgall/status/831893715279036418

I've added a rule, and it appears to work for me. Let me know if it works for you and I'll roll it out across the other languages, too.

diegorodriguezv, 2017-02-15:
Tested several urls in Chrome, Firefox and IE. Also in a couple of Android browsers (in Android 4.0 and 6.01 ). All fine.

closed 2017-03-16 by RogueM