pinout-xyz/Pinout.xyz #197: Second UL in Pi PoE messed up formatting

francesco-vannini, 2017-03-16:
Hi guys,

something has gone wrong with the second unordered list for the Pi PoE. The md file seems OK so it must be something on your production environment I suppose.

RogueM, 2017-03-16:
well, I reckon it's the lack of a newline prior to the list.

we'll fix it soon... ultimately, you should probably check all your PR against our render, the tool chain is explained on the front of the is repo and is what we use to test things out pre-publishing.

closed 2017-03-16 by RogueM

francesco-vannini, 2017-03-16:
OK, sorry guys I imagined the preview within GitHub would have already been
a guarantee that things would be fine on the platform.
Thanks for fixing it.

RogueM, 2017-03-16:
yes, I think it's definitely a bug in python Markdown, but there's also a syntax war going on about the requirement or not to have a leading newline in front of a markdown ul, so hard to tell where to put the finger on that one.