pinout-xyz/Pinout.xyz #198: Zero LiPo doesn't need 5V

PrzemoF, 2017-03-16:
https://pinout.xyz/pinout/zero_lipo#

```
Details
    Made by Pimoroni
    8 pin header
    Needs 5v power
    Uses 1 GPIO pins
    More Information
    Buy Now
```

It doesn't need 5V - it provides 5V stepped up from LiPo battery (3.7V-4.2V)

RogueM, 2017-03-16:
what you say is entirely correct... this is unfortunately a side effect of the Details sections being partly constructed from the pinout.

I think it is pretty safe to assume that anyone using a Zero Lipo understands what is meant here, that the Pi 5V rail need to be connected to the Zero Lipo, for purposes of select wiring.

Granted we could hardcode a way to clear up any misunderstanding, I don't like hardcoding things, though there might be a way to derive a rule from another tag and omit the automatic population on the right hand side in the very few cases that need it... I'll see what can be done.

PrzemoF, 2017-03-16:
Maybe instead "Needs 5V power" use "Needs 5V power connection"? That should cover all "normal" devices and "power supply" devices.

RogueM, 2017-03-17:
I think it's better but only slightly, if a case that the current wording is ambiguous can be made (and I accept the case exist). Strings are also subject to translations, so it is hard to ensure the different languages carry the intend.

I think it should be possible to omit any mention i the details for boards of 'type=power'... I haven't looked at the code yet, hopefully it's not too cumbersome to implement... I'll give it a thought at the weekend. That said I'll keep your rewording in mind if I can't work out a practical way to remove the line where desirable.

RogueM, 2017-03-17:
workaround implemented... will go live in a little while though.

closed 2017-03-17 by RogueM