pinout-xyz/Pinout.xyz #203: 24 DPI is supported on the Pi

deltabeard, 2017-03-30:
According to your [DPI page](https://pinout.xyz/pinout/dpi), the Raspberry Pi only uses up to 6-bits for each colour RGB each. This is incorrect, as the Raspberry Pi can output 24-bit colour using 8-bits for RGB each, and some other colour modes too as written [here](https://www.raspberrypi.org/documentation/hardware/raspberrypi/dpi/).

Additionally, it would be great if pinout.xyz would be able to show all six modes of DPI that the Raspberry Pi can use.

Thanks.

lurch, 2017-03-30:
I suspect the pinout.xyz info is probably based on what was made publicly available when the GertVGA666 was released? (at which time, only one DPI mode was documented).

As you note, more modes have now been documented, and see also http://elinux.org/RPi_BCM2835_GPIOs in case it's helpful.

RogueM, 2017-04-01:
well spotted... and I suspect @lurch is right about why it DPI was documented in the manner it was.

I have now documented the RGB24 mode, which may in practice not be the one most used in the real-world, but arguably the most adequate for pinout to documented (if only one is to be).

... right now documenting all modes neatly and properly would be beyond pinout.xyz's current capability but I will keep this ticket open as an enhancement, so we can possibly return to it in the future.

Gadgetoid, 2019-01-26:
I think it's out of scope- and it's been 2 years ;) Thanks all for your input, but I'll shelve this for now.

Perhaps some better linking to external technical documents that detail this stuff would help?

closed 2019-01-26 by Gadgetoid