pinout-xyz/Pinout.xyz #207: Missing pins on MotoZero

oscarcp, 2017-05-09:
Although the official documentation of the MotoZero says that the board uses 12 of the GPIO pins, this is incorrect, since the ICs use power pins as well and they are not documented anywhere. These are the pins that have to be used in case you don't use the GPIO block header and want your own wiring.

GPIO 39 - Ground (Common)
GPIO 20 - Ground (Common)
GPIO 2 - 5V (Motor 1 & 2)
GPIO 4 - 5V (Motor 3 & 4)

The ground pins need to be both connected for the chips to work, a disconnection in any of them means that the board won't work.

The 5V pins control each of the motor sections, one controls motors 1 & 2, the other 3 & 4, they can be safely disconnected if you want to keep one of the chips deactivated.

RogueM, 2017-05-10:
thanks for info. A very odd choice to not interconnect phys 2 and 4... but hey...

are you saying that phys 20 and 39 are the only relevant ground pins involved, i.e that all others can be safely omitted and only those are relevant to operation? (not that your post leaves room to interpretation, but since I have no access to this board, just to be sure I did not misunderstand the meaning of your report).

oscarcp, 2017-05-10:
Hi @RogueM, indeed it was a weird choice of pins, but I think it's by design.

In terms of the phys 20 and 39, I confirm that they are the only ones required for operation. I actually checked every single pin in the GPIO to make sure. I attach also a picture of my fully functioning board with custom wiring.
[image]

RogueM, 2017-05-10:
thanks, I'll amend.

... not sure an add-on that attaches to the Pi header but separates power can be called a design feature though. Presumably, if it was there would be some way on the board itself to switch off a pair of motors, rather than require select wiring. Odd to say the least.

closed 2017-05-10 by RogueM