pinout-xyz/Pinout.xyz #209: Need signal levels

ramseyrt, 2017-05-17:
Hello,

Could you add BCM signal level information to the GPIO pin-out?  Having the BCM signal level information would help when interconnecting Raspberry Pis with other electronics (such as ham radios).

Thanks!

lurch, 2017-05-17:
What do you mean by "signal level"? All the GPIO pins on the Raspberry Pi (both input and output) use 0V for 'off' and 3.3V for 'on' - if you need anything more you'll need to use some kind of level-shifter.

ramseyrt, 2017-05-17:
Hello Andrew,

Thanks for the response.  I'm trying to connect my Raspberry Pi to my Ham Radio for use with the SoundModem (packet radio application).  In this configuration, I need to provide a Push-to-Talk signal to the radio (typically sent using the RS-232 RTS signal) from my Raspberry Pi.

I've read that the RS-232 RTS signaling uses +/-12v (where +12 means ready to send and -12v means stop sending).  If the Raspberry Pi can only provide +/-3.3v, then it probably couldn't support RTS signaling.

Thanks,

Rob

RogueM, 2017-05-18:
general hardware specifications is besides the purpose of the site. The Raspberry Pi foundation documentation include a very large amount of technical qualifications of their hardware and I highly recommend you check it out:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/gpio/README.md

closed 2017-05-18 by RogueM

lurch, 2017-05-18:
> If the Raspberry Pi can only provide +/-3.3v from the BCM pins

The Raspberry Pi can't provide -3.3V, only 0V and 3.3V.  There are a number of RS-232 interface add-ons available though (e.g. https://www.abelectronics.co.uk/p/75/Serial-Pi-Zero or https://www.amazon.com/dp/B0007T27H8/ ), although I obviously have absolutely no idea whether they're suitable for your purposes or not.

ramseyrt, 2017-05-18:
Hello Andrew,

I saw that my case was closed.  I still think a link to the Pi hardwar
spec, provided by an admin in the case notes, would be helpful for others.
You have a great site and the link would only make your site more useful.

Thanks for the links.  I'll check those out :)

Thanks!

Rob