pinout-xyz/Pinout.xyz #218: pinout.xyz for 96boards

radhikap18, 2017-07-14:
Hi,
I wish to create a pinout website for 96boards and seeing that it is under the creative commons license, I have the pinout.xyz repository forked. I just had a few questions regarding the permissions and the extent to which I am allowed to use, transform and release the content already present in the repository. I have listed out my concerns.
1. Would it be alright to modify the readme file in the main repo according to 96Boards. We plan to have a credits.md file for all the necessary attributions to you and release everything under the same license.
2. Our goal is to create our own upstream so would that seem possible,
3. We plan on having a different url name that would be specifically for 96boards pinout.
Any help would be appreciated. Thank you.

Gadgetoid, 2017-07-17:
1. Absolutely.
2. That would make sense! Although I really feel for you having to tackle my awful Python site generator now!
3. Sure! Go for it.

The Pinout.xyz licensing is a little complex because I opted for a Non-Commercial clause. In retrospect, I can't even clearly define the reasons why I opted for this clause, and it's actually had some negative impact (for example: it prevents some people making contributions due to their employment terms).

I am considering a license change to drop the Non Commercial clause since I think a lot of third party board manufacturers could benefit from a similar site- many, many, many boards use a very similar Pinout to the Raspberry Pi- and the Non Commercial clause may discourage them, despite that not being its intent.

So. TLDR: I fully endorse what you're trying to achieve, and would love to see more Pinout versions serving different ecosystems. If I can make any clarifications to the licensing or README to explicitly allow this sort of thing, and make it clear I support it then I'm all ears!

radhikap18, 2017-07-17:
Thank you so much for such a detailed response! That was more than helpful. I will be in touch if something comes up.

closed 2017-07-17 by radhikap18