pinout-xyz/Pinout.xyz #223: More Pins needed for SenseHat?

mriAppDev, 2017-07-28:
Greetings!

I'm a BIG FAN of your site and have shared it with dozens of other folks too.

My apologies if I've missed something, but when I hooked up the EIGHT Pins indicated on the site for the Sense Hat, the device would not power up (initial "Rainbox" display for a few seconds) when the Pi booted up nor was it available to use from python programs  (Device Not Detected error).

When I connected THREE additional PINS, the Sense Hat began working again.
My goal was to use the BARE MINIMUM number of pins for the Sense Hat so that some others might still be available for other purposes.

I've tested both the marquee messaging and the humidity readings on my hardware.

The pins I added, which are not currently on the Pinout.xyz site are:
Pin: 27, BCM 0 ==> ID_SD
Pin: 28, BCM 1 = ==> ID_SC
A ground Pin ==> I used Pin 39

Please let me know if there's any additional information needed.
I hope this helps improve Pinout.xyz's great work.

My apologies if I've not followed the appropriate procedures or if I've gotten something wrong -- it's my first "issue report" on GitHub.

JC Irvine

Gadgetoid, 2017-08-01:
I think the issue is that Sense HAT is rather unique in its use of the EEPROM- given Raspberry Pi's privileged position, both from a knowledge perspective and from their access to their own repositories, they were able to support Sense HAT at quite a low level.

The upshot of this is that without the EEPROM connected at boot-time the Raspberry Pi doesn't receive the Device-Tree snippets that tell it how to configure the various drivers used for Sense HAT. This, however, doesn't mean that Sense HAT isn't functioning- it just means the drivers aren't loaded.

You can probably omit these pins and, instead, add `dtoverlay=rpi-sense` to your `/boot/config.txt` to accomplish the same thing.

You're right, though, this caveat should probably be documented on Pinout, since it's unique to Sense HAT (as far as I'm aware).

P.S. Thanks for your words of support :D

bsimmo, 2017-10-24:
It is not unique.  It is just following the official HAT specification.  These pins are for anyone who uses an EEPROM for ID.
https://github.com/raspberrypi/hats/blob/master/README.md

I came here to mention the missing pins as they tend to be needed by most people. 
It's useful to know for anyone that uses wires to connect it up or for people using EEPROM to in case it is a problem.
MagPi ran an article a bit back on using them.
https://www.raspberrypi.org/magpi/make-your-own-hat/

bsimmo, 2017-10-24:
MrAppDev, as mentioned above you should be able to get away with not using them if you trigger the dtoverlay yourself.

Also the SPI pins don't need to be connected as they are only for reprogramming the EEPROM and ATTiny88 as far as I'm aware https://www.raspberrypi.org/blog/buy-the-sense-hat-as-seen-in-space/ looking at the schematic, pins19/21/23/22/24 ?

closed 2026-08-19 by Gadgetoid