pinout-xyz/Pinout.xyz #228: Syntax error in example code for the Blinkt! board from Pimoroni

shortlisted, 2017-08-01:
The example code in at Pinout.xyz/src/en/overlay/blinkt.md generates a TypeError when used.

The example code is:
```python
r = randint(255)
g = randint(255)
b = randint(255)
```
But should be:
```python
r = randint(0, 255)
g = randint(0, 255)
b = randint(0, 255)
```

RogueM, 2017-08-03:
what version of Python are you using? while I' sure the code could be made friendly regardless of version, I'm pretty sure the current one can only function in Python3.

shortlisted, 2017-08-03:
That syntax doesn't work in either version. Both python 2.x and 3.x has the following syntax for randint:
```python
random.randint(a, b)
```
At `randint(255)` python2.6.2 will say:
_TypeError: randint() takes exactly 3 arguments (2 given)_
And python3:
_TypeError: randint() missing 1 required positional argument: 'b'_

pjbroad, 2018-12-20:
I had this problem too and came across this issues when I was about to submit my pull request.  I fixed the codes examples exactly as @shortlisted  suggested.

Gadgetoid, 2019-01-26:
Fix merged and published, but it's behind the cache! Will be live soon. Thanks all!

closed 2019-01-26 by Gadgetoid