pinout-xyz/Pinout.xyz #230: Which connector does this refer to?

AndreKR, 2017-08-07:
Here's what I see:
[image]
There are 40 pins listed on pinout.xyz, which connector do they refer to?

Gadgetoid, 2017-08-07:
Pinout predominately documents HATs and pHATs designed for the 40-pin GPIO header on the Pi B+ and upwards.

The 26-pin header on earlier Pi's (such as yours) is a subset of the 40-pin header, which includes all of the top 26 GPIO pins documented, but not BCM 0/1 and those after them.

Most HATs or pHATs that use just the first 26 pins  on the header can be used with your Pi, but you will need a 26-pin stacking header to make them fit.

AndreKR, 2017-08-07:
Ah, thanks, very helpful. You could copy that straight to the website. :)

closed 2017-08-07 by AndreKR