pinout-xyz/Pinout.xyz #245: Broken link on Score:Zero page

lurch, 2017-11-30:
https://pinout.xyz/pinout/scorezero has a `Schematic` link which goes to https://pinout.xyz/pinout/None which doesn't work ;-)

Gadgetoid, 2017-11-30:
D'oh! I didn't catch that one in my "testing"!

On a related note- I've been considering creating some sort of schema or validator to make the process of checking if an overlay file is sane easier. Right now it's:

* Build the whole site
* Wait for some part of the process to break
* Groan, and find what caused it to break
* Dig through the YAML looking for the error
* Fix
* GOTO 0

lurch, 2017-11-30:
That'd be sweet, but it can be a lot of effort to get the schemas set up initially.
A quick google search finds https://stackoverflow.com/questions/5060086/yaml-schema-validation which in turn recommends https://github.com/Grokzen/pykwalify

Gadgetoid, 2018-01-25:
I fixed this with an explicit check for this condition for now- will help catch it in future I hope.

closed 2018-01-25 by Gadgetoid