pinout-xyz/Pinout.xyz #249: Flip option for when looking at the underside?

avivbeeri, 2017-12-05:
I love this tool, but I have a small feature request.

It'd be useful to have an option to toggle if I am looking at it from the top or the bottom, as I find myself soldering to the underside of the Pi on occasion.

lurch, 2017-12-06:
Interesting idea. As a temporary workaround, you could look at your monitor via a mirror ;-)

I wonder if an option of a horizontal layout (in addition to the current vertical layout) would also be useful for some people?

avivbeeri, 2017-12-06:
I would definitely find that useful also.

Gadgetoid, 2017-12-14:
The trouble with a horizontal layout is fitting the text labels.

For our API images, I'm looking at very minimal horizontal images that do not have the labels, like so:

[image]

These might be a little *too* minimal though!

Mirrored is an interesting idea, though. That should be possible with purely javascript/css alone, making it a client-side option.

lurch, 2017-12-14:
@Gadgetoid That image looks neat. Does dark-green indicate "no pin connection"? I wonder if something like this might be clearer?
[image]

Gadgetoid, 2017-12-16:
It's much clearer, in fact, I might go for something in-between. With a very light colour to indicate no-connection.

Gadgetoid, 2017-12-16:
[image]

^ how's this

rgov, 2019-07-07:
I played around with this and simply reversing the `top` and `bottom` CSS classes works to flip the labels. The more difficult thing is flipping the PCB background. For instance, the white "Raspberry Pi Pinout" text would become illegible against a white background.

Gadgetoid, 2026-08-18:
Realised this got added to the Pico pinouts over time, but never made it back here. Will look into it!

Gadgetoid, 2026-08-18:
Added! https://pinout.xyz

closed 2026-08-18 by Gadgetoid