pinout-xyz/Pinout.xyz #251: SenseHAT

Bernd54Albrecht, 2018-01-04:
Thanks for the overall useful information.

I have taken the SenseHAT away from the J8 header to enhance sensor readings. After connection of power and I2C lines   the SenseHAT did not work properly. I recalled that it is a real HAT; therefore I connectet also pins 27 ID_SD and 28 ID_SC. Then, the programs ran as expected, with more realistic temperature readings.
Recommendation: Mark the physical pins 27 and 28 as necessary / "in use"

Gadgetoid, 2018-01-16:
I've been planning this for a while, since it affects what HATs might be compatible with each other. We're already storing whether or not HATs have an EEPROM, so it should be quite simple to convey that information somehow.

Incidentally, in the case of Sense HAT they're not actually *required* but it's quick and easy to hook them up, versus making sure the correct drivers run on startup.

IIRC you can disconnect those pins and put `dtoverlay=rpi-sense` into `/boot/config.txt` and it'll work just like they were connected.

lurch, 2019-10-10:
Duplicate of #223 ?

closed 2026-08-19 by Gadgetoid