pinout-xyz/Pinout.xyz #252: HAT Explorer 700 / Pioneer 600

Bernd54Albrecht, 2018-01-04:
I am looking for more information on the HATs Joy-it Explorer 700 or Seeed Pioneer 600.
Request inclusion in the HAT section.

Gadgetoid, 2018-01-23:
This Pioneer 600? - https://www.seeedstudio.com/Raspberry-Pi-Pioneer600-p-2771.html

It looks like a Waveshare board. I wonder why is also goes by the name Joy-it Explorer 700. Any ideas?

I'll see what I can do!

Bernd54Albrecht, 2018-01-24:
> 2771.html

Hi,

no idea, why the waveshare is called 600 and Joy-it 700.

In the meantime, I sent the part back, because some features did not work
properly. Nevertheless, an interesting alternative to the SenseHAT.

My new toy is a GSM / GPS HAT (see link below). Pi Zero used as Telefone
incl. SMS and Position/Height/Course and Speed from GPS/GLONASS. We will
send it to the stratosphere on 3rd March.
And again: poor documentation - a case for you. By the way, it uses only
UART, it can be combined with SenseHAT and extra long header.

https://www.waveshare.com/wiki/GSM/GPRS/GNSS_HAT

Greetings
Bernd Albrecht

closed 2018-04-08 by Gadgetoid