pinout-xyz/Pinout.xyz #253: Add SparkFun Pi Servo HAT Board

Patronics, 2018-01-05:
And Ideally any other HATs not listed that are carried by SparkFun. [Here's the Servo HAT](https://www.sparkfun.com/products/14328),[its schematic](https://cdn.sparkfun.com/assets/1/a/1/6/3/PI_Servo_Shield_v10.pdf), and [the other HATs they carry](https://www.sparkfun.com/categories/396)

Gadgetoid, 2018-01-23:
I absolutely agree! Sparkfun are a fantastic brand - I was reading their business journey blog recently and have, of course, used a few of their products :D - I'd love to bring their stuff aboard.

I'll earmark some Patreon funds to pick a few up and get them added.

Any

Gadgetoid, 2018-01-24:
Started: https://pinout.xyz/boards#manufacturer=SparkFun

Good documentation for that board- and it's very close to an AB Electronics one, so low-hanging fruit. Onto the next one!

closed 2018-01-25 by Gadgetoid