pinout-xyz/Pinout.xyz #262: Add default pull up/down states to RPi GPIO table

dougcooper, 2018-02-25:
It would be helpful to know the default pull up/down states of the GPIO which aides in making decisions on which pins to wire to for a given circuit. Here's a link to the reference manual https://web.stanford.edu/class/cs140e/docs/BCM2837-ARM-Peripherals.pdf. Info starts on page 102.

lurch, 2018-02-25:
If you want all the gory low-level details (and with errata applied - AFAIK that "BCM2837-ARM-Peripherals.pdf" is entirely unofficial) see https://elinux.org/RPi_BCM2835_GPIOs

Gadgetoid, 2018-03-13:
I'm not sure the default pull states can be relied upon when using things with HAT EEPROMs?

Anyway, @lurch is correct- this level of detail is beyond the scope of Pinout.xyz. Although maybe we could build and host a prettified version of this gory table ;)

dougcooper, 2018-03-13:
The only reason I proposed inclusion is because you are basically replicating the tables in the BCM data sheet / elinux table, but omitting the only other piece of relevant info - the default GPIO state. Agreed that they can’t apply when hats are used.

lurch, 2018-03-13:
If the default GPIO pull-states are critical to your circuit, you _can_ also override them using [dt-blob.bin](https://www.raspberrypi.org/documentation/configuration/pin-configuration.md) .

dougcooper, 2018-03-17:
Hey fellas. Please do continue to correct me if you think I’m wrong, but I don’t think it’s actually possible to overwrite the default state of the pins at power up. What you are describing is the kernel enumerating the device tree after the boot loaders have done their job. Unfortunately the GPIO pin initial states are defined in hardware and cannot be overwritten until some piece of software is loaded into memory.

lurch, 2018-03-17:
You're correct that "the GPIO pin initial states are defined in hardware and cannot be overwritten until some piece of software is loaded into memory" but `dt-blob.bin` is actually _different_ from DeviceTree (despite the confusing name) and contains settings that get applied by the boot firmware, before the Linux kernel even starts loading (i.e. the same boot firmware that also reads `config.txt`).

If the initial pull-states are critical to your circuit, another option would be to simply use external pull-up / pull-down resistors, which will override the weak internal pull-ups / pull-downs.

lurch, 2018-05-24:
Related: https://github.com/raspberrypi/documentation/pull/904 (and the forum thread that it links to)

dougcooper, 2018-05-24:
> Related: raspberrypi/documentation#904
> <https://github.com/raspberrypi/documentation/pull/904> (and the forum
> thread that it links to)

So you “could” add external pull-ups I suppose. Except now you have to add
unnecessary parts to your design instead of taking advantage of what the
chip offers. You guys are the moderators so if you strongly disagree with
my recommendation to add the states then by all mean close this issue.

closed 2018-05-26 by dougcooper