pinout-xyz/Pinout.xyz #267: Where is one

ziuq, 2018-04-05:
I think it worth while to note that Pin #1  has a square solder pad  and, possibly, to adjust the graphics.   The graphic indicates the board has corners just after both mounting screw holes.  I know this is incorrect for the 3B, and I think it is for the entire line.

Thanks for the nice pin out pages.

ziuq, 2018-04-06:
> For me, Pin 1 _does_ get drawn with a square 'solder pad' - what OS and browser are you using?
> [image]
>
> The rounded-corners are correct for the Model A+ and the Pi Zero :wink:

I guessed the older versions likely had a different format.

Yes, my chromium shows the square pad.  I did not mean that your page is
inaccurate, rather that it is possible to be confused which side of the
the board to look at. A board already mounted may hide the pads. 
The beginner may not know the square is significant.  The 3B board does
not does not label pin 1 that I can see, but for the square pad.
All these seem to call for a comment or something.

Really impressive page.  Thanks,

Rob

lurch, 2018-04-06:
I ought to clarify that it's not **my** page, I just help out with small tweaks occasionally :slightly_smiling_face:

Gadgetoid, 2018-04-08:
I see what you're getting at, but I haven't come across many people who are confused by the layout. Perhaps I just never heard from them? Still I'm sure someone else has raised similar concerns, because the skeuomorphic design I opted for doesn't accurately represent the Pi 3, or any larger form factor boards.

loranbriggs, 2018-05-01:
I agree with ziuq, it may possible to get the diagram 180 degrees off. I like the simple design but it may be helpful to reference which direction the other ports like USB or HDMI are. Just offering constructive feedback, love the site otherwise. I could even contribute with a pull request later this week.

Gadgetoid, 2019-01-26:
Pinout's long overdue a minor design overhaul- particularly to make it mobile-friendly - so I'm all ears.

Perhaps we can stand to gain some extra horizontal width, or use the width available in wider browser windows, to illustrate the HDMI ports or other visible features of the Pi for easier orientation.

A quick-fix might be to extend the left-hand pinout menu down and squeeze in some simple blocks to represent the two USB ports- but then I worry about A+ and Zero form-factors.

Another user has suggested I add a visual "key" to indicate the meaning behind the highlights pins, too, so there are a few things to consider.

rgov, 2019-01-28:
I came here to file the same issue because I was confused by the orientation. When I got a Pi 3B, which came with the GPIO header pins pre-soldered, I immediately placed it in an opaque case, so I cannot see the square solder pad.

But I notice that the white outline around the pins, as in this photo, has a notch at pin 1, I think:

[image]

tuftii, 2019-06-11:
FWIW I use the green coloured block to the left of the pins. I know that if the majority of my Pi is over white webpage then it's the wrong way round...

But then I'm an odd one and I'm not sure that's a common way of looking at it :woman_shrugging:

lurch, 2019-06-11:
Luckily, whilst there are Pis with Green, Blue and Red circuit boards, there aren't any Pis with white circuit boards :+1: