pinout-xyz/Pinout.xyz #268: RUN function missing

kllsamui, 2018-05-01:
[https://pinout.xyz/pinout/pin5_gpio3#](url)
i miss the info about BOOT 
when pin5 and pin6 have a push button.
( and also shutdown function with add config in
/boot/config.txt
dtoverlay=gpio-shutdown
)

Gadgetoid, 2018-05-12:
Is this exclusively the run header, not the main GPIO?

kllsamui, 2018-05-12:
sorry, i was not talking about [RUN] [PEN] and RPI3B+ recent change,
what makes it difficult.
i ref to
[https://www.google.com/search?hl=en&as_q=&as_epq=gpio+shutdown&as_oq=&as_eq=&as_nlo=&as_nhi=&lr=&cr=&as_qdr=all&as_sitesearch=https%3A%2F%2Fgithub.com%2Fraspberrypi%2Ffirmware%2Fblob%2Fmaster%2Fboot%2Foverlays%2FREADME&as_occt=any&safe=images&as_filetype=&as_rights=](url)