pinout-xyz/Pinout.xyz #277: 404 for "More Information" and "Buy Now" links for Ryanteck boards

claremacrae, 2018-06-10:
From these two pages:
* https://pinout.xyz/pinout/rtk_motor_controller#
* https://pinout.xyz/pinout/traffic_hat#

clicking the "More Information" and "Buy Now" links gives 404s, understandably.

Maybe remove the links, if possible?

claremacrae, 2018-06-10:
I was going to just edit this in the github web UI, but I see there are lots of translations all referring to the same URLs, which would also want to be updated...

claremacrae, 2018-06-10:
There are references to ryanteck.uk in various other pages, such as buying the RasPIO Duino

lurch, 2018-06-11:
The RasPiO Duino links could probably be changed to https://shop.rasp.io/collections/raspio-collection/products/raspio-duino-low-cost-easy-way-into-arduino-programming-on-the-raspberry-pi ?
ping @raspitv

Gadgetoid, 2018-06-12:
Oh d'oh! I totally didn't click when Ryantech was shut down- these links must have been broken for ages! Thanks for pointing them out.