pinout-xyz/Pinout.xyz #293: Boards TODO

Gadgetoid, 2018-10-10:
In an effort to keep track of HATs, pHATs and Rasperry Pi add-ons which have not yet been documented on Pinout.xyz I'm compiling a - probably not - comprehensive list here. This list should give me (and others, if you wish to contribute) a working list of boards which need to be documented.

## Suggest a board
Please comment below to suggest a board and I'll add it to the list.

## Contribute a board
If you're planning to contribute an image/overlay for a board in the list below, let me know so I can mark it as being worked upon.

# Board list

## SparkFun
* ~AVR Programmer HAT~ - https://pinout.xyz/pinout/sparkfun_avr_programmer_hat
* LCD Touchscreen HAT
* Qwiic HAT
* ~Pi Servo HAT~ - https://pinout.xyz/pinout/pi_servo_hat
* MOTORplate ?? - These plates are hooge! Will be tricky to prep images for, but they fit the 40pin header
* RELAYplate ??
* DAQCplate ??

## Pololu
* ~Dual G2 High-power Motor Driver~ - https://pinout.xyz/pinout/dual_g2_high_power_motor_driver
* A-Star 32U4 Robot Controller LV with Raspberry Pi Bridge
* A-Star 32U4 Robot Controller SV with Raspberry Pi Bridge
* Dual TB9051FTG Motor Driver
* Dual MC33926 Motor Driver
* Dual MAX14870 Motor Driver
* Dual DRV8835 Motor Driver
* Romi 32U4 Control Board ?? - HOOOGE!!

## Waveshare
* 250x122 E-Ink display HAT
* 264x176 E-Ink display HAT
* 4G/3G/2G/GSM/GPRS/GNSS HAT
* High Precision AD/DA Expansion Board
* 2.8" LCD
* Relay Board
* GSM/GPRS/GNSS/Bluetooth HAT
* 128x128 1.44inch LCD Display HAT
* Pioneer 600 - I can't remember what became of this, requested here: https://github.com/Gadgetoid/Pinout.xyz/issues/266
* 3.5" LCD 320x480
* BN-IoT / eMTC / EDGE / GPRS / GNSS HAT
* 212x104 E-Ink display HAT
* 240x240 1.3" IPS LCD display HAT
* Servo Driver HAT
* Motor Driver HAT
* RS485 CAN HAT
* 128x64 1.3" OLED display HAT
* Motor Driver Board

scotty3785, 2018-11-08:
The new Raspberry Pi TV uHat isn't documented yet either.

Gadgetoid, 2026-08-19:
I'm relying on manufacturers/vendors or interested third parties to submit boards, since they have a strong incentive to do so and I have neither the time nor the resources!

closed 2026-08-19 by Gadgetoid