pinout-xyz/Pinout.xyz #301: Add pi-desktop pinout

TiceRex, 2019-01-10:
I would like to suggest a board as new addition: Element14's Pi-Desktop.
Pinout can be found here:
https://github.com/pi-desktop/pi-desktop/blob/master/Pins-used-by-pi-desktop/Pins-used-by-pi-desktop.md

lurch, 2019-01-10:
Hmmm.... https://github.com/pi-desktop/pi-desktop/issues/4

TiceRex, 2019-01-11:
Seeing the "unofficial" pi-desktop software here: https://github.com/hoopsurfer/pidesktop, that blinking is actually the shutdown timer.
The power management communication between the RPi and the MCU is done through GPIO.22 and GPIO.23
The RTC communication is done on I2C, and can't see anything on BCM17.
I can't find any reference to GPIO.24 (BCM19) and GPIO.25 (BCM26) neither, as mentioned in the original doc.

So, It seems to be not that easy. :-(

lurch, 2019-01-11:
> So, It seems to be not that easy. :-(

Perhaps this is one of those situations where there's multiple (unannounced) revisions of the board, and different revisions use different pins? :man_shrugging: (just to clarify, I don't have a Pi-Desktop)

TiceRex, 2019-01-11:
I don't think so. The original software would not work with different pinouts.
On the other hand the factory software is very unreliable (could not handle the reboot process), which _could_ confirm some kind of change. Probably this is why Element14 also refers to the unofficial software on their website (see Application Library under Downloads).
https://www.element14.com/community/docs/DOC-83477/l/desktop-computer-kit-with-expansion-board-that-can-turn-a-raspberry-pi-into-a-real-desktop-pc

closed 2026-08-18 by Gadgetoid