pinout-xyz/Pinout.xyz #306: Add pinout for PecanPi DAC

orchardaudio, 2019-03-03:
Hello,

Please add the pin out for my company's upcoming product the PecanPi DAC.

Pin out is show in image below:

Please let me know if you need any additional information.

Thanks;
Leo

[image]

lurch, 2019-03-03:
If you have a look at e.g. https://pinout.xyz/pinout/dac_hat you'll see the kind of info typically required... (description, photo, product URL, etc.)

orchardaudio, 2019-03-03:
Thank you. I will provide the additional items.

Regards,
Leonid (Leo) Ayzenshtat
Orchard Audio
How do you like them apples?
http://www.orchardaudio.com
http://www.facebook.com/orchardaudio
http://www.twitter.com/orchardaudio
http://www.instagram.com/orchardaudio

orchardaudio, 2019-03-05:
> If you have a look at e.g. https://pinout.xyz/pinout/dac_hat you'll see the kind of info typically required... (description, photo, product URL, etc.)

Here is the additional info:

PecanPi™ DAC
The PecanPi™ DAC is an ultra high performance I2S sound card for the Raspberry Pi. It features both fully balanced (XLR) and single ended (RCA) outputs as well as a built in high performance headphone amplifier. The I2S clocks are fully re-clocked using a Crystek femtosecond jitter oscillator.

Specifications:
- Balanced output voltage: 5Vrms (XLR)
- Balanced output impedance: 200 ohms
- Single ended output voltage: 2.5Vrms (RCA)
- Single ended output impedance: 0.05 ohms
- Peak power into 32 ohm headphones: 390mW
- Peak power into 16 ohm headphones: 780mW
- Supported Sample Rates: 44.1, 48, 88.2, 96, 176.4 and 192kHz
- Supported Bit Rates: 16 and 24-bit
- All Raspberry Pi GPIO accessible on header
- HW volume control available with potentiometer

Details:
- Made by Orchard Audio
- Needs 9V Power Supply
- Can back-power the Raspberry Pi for single power supply operation.
- [More Information](http://www.orchardaudio.com/pecanpi)

Here is the picture:
![img_0344_1](https://user-images.githubusercontent.com/34345446/53772705-6968f680-3eb5-11e9-88f7-5c108ea4d1d8.jpg
)

Please let me know if that is sufficient or if you will need additional info.

monglet, 2019-04-13:
@orchardaudio Needs a better picture... 

> Note that as part of the submission, a top-down view of the board in the form of a png is expected. If you can't produce the png file yourself, just duplicate and rename template.png but make sure to include a url somewhere in the overlay where we can fetch a suitable graphic.

Look at the https://pinout.xyz/boards page for how your board photo needs to appear...

orchardaudio, 2019-04-16:
I will work on getting you a png per your requirements.

orchardaudio, 2019-05-01:
> @orchardaudio Needs a better picture...
>
>
> Look at the https://pinout.xyz/boards page for how your board photo needs to appear...

Here is the PNG image.
[image]

orchardaudio, 2019-06-19:
Hello, any update on this?

lurch, 2019-06-19:
I just spotted that http://www.orchardaudio.com/pecanpi says "PecanPi Steamer", which is a typo you might want to fix :wink:

orchardaudio, 2019-07-20:
> I just spotted that http://www.orchardaudio.com/pecanpi says "PecanPi Steamer", which is a typo you might want to fix 😉

Fixed

orchardaudio, 2019-07-20:
Any update on when this will get added to pinout.xyz

orchardaudio, 2020-08-27:
Has this been added to pinout.xyz? Is there other info that you need?

orchardaudio, 2021-12-07:
Hello. Has this been added to Pinout.xyz? Is there some other info needed?

orchardaudio, 2022-10-25:
> From a customer of Orchard Audio who has a PecanPi DAC attached to a Raspberry Pi and is looking for its pinout.
> Having the pinout available at pinout.xyz would be a lot more accessable than just having it shown in this thread which is the only source I have found for it.
> Is there a reason it has not been added to Pinout.xyz?

Hello,

I have tried to add the pinout to your site, but you never got back to me. See below.

Regards;
Leonid (Leo) Ayzenshtat
Orchard Audio
M  +1 504-233-3444
E  leo@@orchardaudio.com
W www.orchardaudio.com
[Rating stars]<https://admin.trustindex.io/logc.php?wid=e287b7e796828580105b66346e>
[Facebook-icon-01]<https://www.facebook.com/orchardaudio> [twitter] <https://twitter.com/orchardaudio>  [instagram_PNG10] <https://www.instagram.com/orchardaudio>  [linkedin-logo-copy] <https://www.linkedin.com/company/25168462/>

***@***.***

closed 2026-08-18 by Gadgetoid