pinout-xyz/Pinout.xyz #308: Needs legend for color codes

mok0, 2019-03-21:
I really miss a legend for the color codes used in the pinout diagram (on the left). I realize the green circles mean that the pin is available, but what are the red, violet, purple, blue and light blue circles? It's not at all clear for a beginner, and it would be very useful with a legend. There is whitespace below the pinout picture where such a legend could be conveniently placed.

lurch, 2019-03-21:
I did a bit of clicking-around, and it _seems_ like:
- black is [ground](https://pinout.xyz/pinout/ground#)
- red is [5V power](https://pinout.xyz/pinout/pin2_5v_power#)
- orange is [3.3V power](https://pinout.xyz/pinout/pin1_3v3_power#)
- blue is [I2C](https://pinout.xyz/pinout/i2c#)
- violet is [UART](https://pinout.xyz/pinout/uart#)
- pink is [SPI](https://pinout.xyz/pinout/spi#)
- green is "anything else"

@Gadgetoid Why are some of the SPI pins coloured green instead of pink?

mok0, 2019-03-22:
Thanks @lurch, that's informative, and IMO that info belongs in a legend, perhaps with the little circles instead of red, orange, etc. So you don't need to come here in order to understand what the colors mean :-)

Gadgetoid, 2019-04-13:
This is good feedback- thank you! In retrospect it seems so obvious that a key should accompany the colours, but it never occured to me since I - understandably - know what they correspond to. D'oh!

Gadgetoid, 2019-04-13:
Deployed to the front page, bottom left under the GPIO - https://pinout.xyz/

Just struck me that I should probably link these to the relevant protocol pages, so I'll keep this issue open for the next sprint.

lurch, 2019-04-13:
Maybe it's just me, but the text in brackets in the new legend looks _awfully_ small :confused: 

> @Gadgetoid Why are some of the SPI pins coloured green instead of pink?

To elaborate: on https://pinout.xyz/pinout/spi it says that BCM 17 is SPI1 CE1, BCM 18 is SPI1 CE0 and BCM 16 is SPI1 CE2, yet those pins are drawn in green and not pink?

lurch, 2019-05-29:
Also, the legend for 3v3 says "3v (Power)" when it should probably say "3.3v (Power)".
And to be pedantic, the 'v' in both the "3v (Power)" and "5v (Power)" legends ought to be in upper-case :wink:

Gadgetoid, 2019-10-09:
Lowercase "v" is cuter 😆 

You're right, though, the text in backets is *very* small and could probably stand to be larger.

Pinout is definitely plaguaed by a few size/space formatting issues that could probably use some more modern CSS to tidy up. @peterblazejewicz's contributions have been great!

Whew it's taken me just over 2 hours to catch up with PRs, fix some issues, do some code tweaks, publish and tag up the repo with a build ID matching the live one. I was supposed to be in bed 😴

Gadgetoid, 2020-08-20:
Okay, the Legend is now legendary and with the addition of the orientation graphic has become the de-facto way to clarify what people are seeing in the Pinout.

Thank you *very* much to all involved, with special emphasis on @lurch without whom I would be swamped under the sheer magnitude of all of this stuff and - despite our terrible disagreement over capitalisation of the "v" - whose input I value and appreciate highly!

closed 2020-08-20 by Gadgetoid