pinout-xyz/Pinout.xyz #309: Correct pinout for AIY Voice Bonnet

hiro345g, 2019-04-11:
AIY Projects V2 Bonnet hardware pinout 

AIY Projects pinout: https://aiyprojects.readthedocs.io/en/latest/voice.html#pinout-40-pin-header

```
       3.3V --> 1    2 <-- 5V
                3    4 <-- 5V
                5    6 <-- GND
                7    8
        GND --> 9   10
                11  12 <-- I2S_BCLK
                13  14 <-- GND
                15  16 <-- BUTTON_GPIO (GPIO_23)
       3.3V --> 17  18
                19  20 <-- GND
                21  22 <-- LED_GPIO (GPIO_25)
                23  24
        GND --> 25  26
     ID_SDA --> 27  28 <-- ID_SCL
                29  30 <-- GND
                31  32
                33  34 <-- GND
  I2S_LRCLK --> 35  36 <-- AMP_ENABLE
                37  38 <-- I2S_DIN
        GND --> 39  40 <-- I2S_DOUT
```
Pinout.xyz pinout: https://pinout.xyz/pinout/aiy_voice_bonnet is not match.

lurch, 2019-04-11:
Ping @scottamain as he contributed the AIY Voice Bonnet page.

Gadgetoid, 2019-04-13:
Looks like we're missing the LED_GPIO, the AMP_ENABLE and - potentially - the extra 3.3v power supply?

Can skip the ID pins as we don't document those.

closed 2026-08-20 by Gadgetoid