pinout-xyz/Pinout.xyz #310: Errors: curl -sS https://get.pimoroni.com/explorerhat | bash

daveyostcom, 2019-04-11:
[This page](https://pinout.xyz/pinout/explorer_hat_pro) says to run this:
 `curl -sS https://get.pimoroni.com/explorerhat | bash`
When I run it, I get:
```
Updating apt indexes...
...W: http://raspbian.raspberrypi.org/raspbian/dists/stretch/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/microsoft.asc.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: http://archive.raspberrypi.org/debian/dists/stretch/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/microsoft.asc.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: https://packages.microsoft.com/debian/9/prod/dists/stretch/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/microsoft.asc.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: GPG error: https://packages.microsoft.com/debian/9/prod stretch InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY EB3E94ADBE1229CF
W: The repository 'https://packages.microsoft.com/debian/9/prod stretch InRelease' is not signed.
```

Linux pi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

lurch, 2019-04-11:
AFAIK the `W: ` signifies that they're just warnings, and so can be ignored? Do you get any other errors? Has the explorerhat software been installed?

JoshuaLowe1002, 2019-04-30:
Hey @daveyostcom you're best submitting an issue over at https://github.com/pimoroni/explorer-hat/issues for issues with the installer. 

However, like @lurch suggests, these look like just warning and not something that should affect the installer hopefully. Maybe try running `sudo apt update` to see if that fixes the warnings as they are apt related warnings.

closed 2019-05-23 by daveyostcom