pinout-xyz/Pinout.xyz #312: Zero2Go Omini Pinout Inaccurate

Exscotticus, 2019-04-15:
I'm pretty sure the [Zero2Go](https://pinout.xyz/pinout/zero2go) Omini pinout is inaccurate.

From the docs...

"Zero2Go Omini uses GPIO-4 (BCM naming) as SWITCH signal and GPIO-17 (BCM naming) as System-Up signal."

"Zero2Go Omini uses an ATtiny841 microcontroller to control the DC-DC converter engines, measure voltages and communicate with Raspberry Pi via I2C interface."

lurch, 2019-04-15:
Looks like https://pinout.xyz/pinout/zero2go is documenting http://www.uugear.com/product/zero2go/ whereas http://www.uugear.com/product/zero2go-omini-wide-input-range-multi-channel-power-supply-for-raspberry-pi/ is a different product? 
Ping @uugear

uugear, 2019-04-15:
@lurch You are right. Zero2Go (the first generation) and Zero2Go Omini are different products. We didn't have time to add pinout for Zero2Go Omini yet.

Exscotticus, 2019-04-15:
Ah OK my bad. So pinout is probably accurate for 1st gen. I'll close this issue.

closed 2019-04-15 by Exscotticus