pinout-xyz/Pinout.xyz #314: Combined "power" pages?

lurch, 2019-05-29:
While looking again at #308 I noticed that all the Ground pins share a common page ( https://pinout.xyz/pinout/ground ) and yet the two 5V pins have separate pages ( https://pinout.xyz/pinout/pin2_5v_power and https://pinout.xyz/pinout/pin4_5v_power ) and the two 3V3 pins have separate pages ( https://pinout.xyz/pinout/pin1_3v3_power and https://pinout.xyz/pinout/pin17_3v3_power ).
Perhaps it makes sense to just have a single page for each voltage? (i.e. https://pinout.xyz/pinout/3v3_power and https://pinout.xyz/pinout/5v_power )

Gadgetoid, 2019-06-08:
Agreed! I'll have to remember how the heck I made the Ground page work and copy that functionality over. The dangling pin pages can be 301d to the result.

lurch, 2020-08-19:
In the latest version of pinout.xyz the 3V3 and 5V pages are successfully combined when clicking on any of the 3V3 or 5V pins on the header, but the 5V and 3.3V links in the legend below the header still go to the old pin-specific pages.

Gadgetoid, 2020-08-19:
Good catch! Fixed in git- pending publish! Thanks

closed 2020-08-20 by Gadgetoid