pinout-xyz/Pinout.xyz #317: Layout oddness on main board page

tuftii, 2019-06-03:
On the live site at the moment there seems to be a 3 board gap between the Cirrus Logic Audio Card and the cube:bit Base board.

[image]

([see here](https://pinout.xyz/boards#))

I'm using Firefox 67.0 with ad-block disabled and no other add ons.

Not sure if this is important or not, or how to solve it, but couldn't see an open issue so here we are :smile:

lurch, 2019-06-03:
I'm seeing the same thing (also with the same version of FF).

tuftii, 2019-06-03:
Oddly, it's not a problem in Chrome. :woman_shrugging: 

[image]

Gadgetoid, 2019-06-08:
Weirdly I don't see this on Firefox! It'll be the text busting out of the bottom of the layout and causing the elements underneath to wrap. There's probably a much more modern way to lay out a grid of elements like this that I should be looking into.

I think the boards page needs some TLC anyway because I *hate* the dependency on Javascript. I've taken pains to make sure there's as little JS and other overhead in this site as possible- Google Analytics notwithstanding for my own vanity.

Giving each manufacturer, form-factor and type its own HTML page would be a nice start perhaps? Ideas welcome!

lurch, 2019-06-09:
> It'll be the text busting out of the bottom of the layout

Oh yeah - in @tuftii 's FF screenshot the text under the Cellular IOT HAT has split over 3 lines, whereas in the Chrome screenshot it's only wrapped onto two lines.

:+1: for reduction of the amount of JS! IMHO the "proper" solution would be to move the current JS filtering logic to the server-side (using something like PHP), but it's up to @Gadgetoid to come up with a solution that meets his available development time and his server's capabilities :slightly_smiling_face:

tuftii, 2019-06-11:
I mean... the simple solution is to just shorten the name as that is *quite* the name for a HAT. Limit to two lines (whatever characters fit roughly?).

lurch, 2019-06-11:
Yeah, editing the name down to just "Cellular IoT Application HAT" probably makes sense. I had a click through to the Sixfab site, and they have a [Cellular IoT HAT](https://sixfab.com/product/raspberry-pi-lte-m-nb-iot-egprs-cellular-hat/) and a [Cellular IoT Application HAT](https://sixfab.com/product/raspberry-pi-cellular-iot-application-hat/) so there's probably no need for the "LTE-M & NB-IoT & eGPRS" suffix? ping @saucompeng

lurch, 2019-06-11:
It's a good thing the JamHAT isn't named the "ModMyPi Buttons and LEDs and Buzzer addon board for Raspberry Jams" :rofl:

tuftii, 2019-06-11:
> 
> 
> It's a good thing the JamHAT isn't named the "ModMyPi Buttons and LEDs and Buzzer addon board for Raspberry Jams" rofl

100% can be arranged, I'm building a PR ASAP... :rofl:

saucompeng, 2019-06-12:
> Yeah, editing the name down to just "Cellular IoT Application HAT" probably makes sense. I had a click through to the Sixfab site, and they have a [Cellular IoT [...]

@lurch  Yes, you are right. Could you fix it?

lurch, 2019-06-12:
> @lurch Yes, you are right. Could you fix it?

I can't, but @Gadgetoid can :wink:

tuftii, 2019-10-01:
I've shortened the text so this formatting issue never happens, but I can now see PR #333 fixes it properly! :+1:

PR #336 (2019-10-09): Update sixfab-cellular-iot-application-hat.md
Shorten name to fix tiling issues in Firefox. Fixes #317.
closed 2019-10-09 by Gadgetoid