pinout-xyz/Pinout.xyz #319: Non-translated "localised" boards pages are missing

lurch, 2019-06-09:
Previously, if a translation of a board page was missing, you just used the English version, e.g. https://de.pinout.xyz/pinout/1_wire_pi_plus#
But it seems there are now some missing pages, e.g. https://de.pinout.xyz/pinout/jamhat#

Which means that https://de.pinout.xyz/boards# has many fewer boards listed than https://pinout.xyz/boards# and for some reason the DE boards page has an empty blue square in the "Type" list on the left :confused:

Gadgetoid, 2019-06-09:
I haven't regenerated the non-UK sites for a little while- mostly out of subtle paranoia. If I break something, I don't want to break everything at once. 

I need to bite the bullet and hit the "Make website" button.

Gadgetoid, 2019-06-09:
Oh boy this is a can of worms I've just opened!

tuftii, 2019-06-10:
Ah. I was about to open an issue noting that some pages seem to be in German today, but I'm assuming this is related?

https://pinout.xyz/pinout/scorezero# for example... :de:

Gadgetoid, 2019-06-14:
In my infinite wisdom I had a Makefile which took a LANG and a TARGET parameter and in almost all cases these should both be the same, but they both default to `en`.

I then did a `make LANG=de` and you can guess the rest 🤦‍♂

tuftii, 2019-06-17:
Genau! :laughing:

Gadgetoid, 2019-10-09:
This reminded me to republish everything :D

closed 2019-10-09 by Gadgetoid