pinout-xyz/Pinout.xyz #320: Add an orientation hint to pinout

pelikhan, 2019-06-10:
There is an ambiguity on this image https://pinout.xyz/resources/raspberry-pi-pinout.png since it does not specify which way it is oriented w.r.t., to the Pi itself.

lurch, 2019-06-10:
See also #267 :wink:

Gadgetoid, 2019-06-14:
Agreed! It's funny how you don't think about this stuff when you look at GPIO headers all day.

pelikhan, 2019-06-14:
I ended writing a game to assist with wiring: https://makecode.com/_2miFR8UfkTob

Gadgetoid, 2019-06-14:
I guess this begs the question- does anyone have any good ideas for hinting at orientation while keeping the minimal look and feel?

I also need to regenerate all these images!

pelikhan, 2019-06-14:
See the board as in  
[image] helped for me.

Gadgetoid, 2019-06-17:
This is interesting, because we've had a number of comment suggesting that even this layout is ambiguous about header orientation. I've always failed to see how, but of course I can't see the wood for the trees!

*These* are *especially* ambiguous:

https://api.pinout.xyz/v1/img/1_wire_pi_plus.png

cst0, 2020-08-16:
Sorry for hopping onto this thread late, but since it's open-- I find this to be ambiguous because the graphic does not indicate the relative location of other landmarks on the Pi, which leaves it open to interpretation. A particular point of confusion is in how the above graphic has the edge of the board wrapping around the mounting hole at the bottom. This is illustrative of the Pi Zero, but not of other Pi devices. A potential additional confusion is that it does not indicate if we should be looking at the top or the bottom of the board: top is the safest assumption, but using the square pad as an indicator has you looking at the bottom of the board. Users will figure it out eventually, but given the goal of creating as easy-to-use a resource as possible, that's potentially problematic.

I can think of two potential solutions that are in-line with the current layout of the page: one would be to simply add an 'orientation' section to the right-hand text, which states that you are looking at the top of the board with the SD card pointing to the top of the page. I've created a pull request (#371) which illustrates this in English. A second solution would be to add other landmarks to the board: three grey rectangles at the bottom would outline the location of the Ethernet and USB ports, but excludes the Pi Zero, so I'm not sure the best way to get around that.

Gadgetoid, 2020-08-17:
I would be tempted to drop the mounting holes altogether in favour of your orientation text. Probably drop the partial green background, too. Less visual noise. This might also free up some space right underneath the pinout to fit this orientation text. It would seem logical that it is part of the "Legend."

Perhaps something drastic like this:

[image]

cst0, 2020-08-17:
I like that location for the text a lot better, it makes the information more accessible (especially on mobile). I think that's a pretty good solution.

lurch, 2020-08-18:
I'm not entirely sure that "with the GPIO on the right" is entirely unambiguous, especially with the removal of the green "edge of the board" background; perhaps expanding it slightly to "with the GPIO on the right and the HDMI port on the left" makes it clearer? :man_shrugging: 

> A particular point of confusion is in how the above graphic has the edge of the board wrapping around the mounting hole at the bottom. This is illustrative of the Pi Zero, but not of other Pi devices.

The Pi Zero, and also the A+ and 3A+ :wink:

Gadgetoid, 2020-08-19:
I've steamrolled a new CSS and text on to try it on for size, might need some refinement.

May perhaps need some kind of reference for orientation so we have the best of both worlds, without the business of the pinout *being* the orientation referenece.

Some ideas:
[image]

Or greeeen!
[image]

Gadgetoid, 2020-08-20:
Okay the orientation key is now live. I'd appreciate any comments/concerns although I'm not likely to make any *drastic* changes in the near future... I'll let it bed in and see what people think.

The BCM numbering image could definitely use some TLC- I'm not quite sure how to communicate the same orientation information in that one.

lurch, 2020-08-20:
Having now seen it in use on the live site, I think perhaps the diagram might be clearer if the two Zero-only mounting holes (in the middle of the board) were removed? :shrug: (or at least have them in a lighter shade or something, than the outer mounting-holes?)

closed 2020-09-18 by Gadgetoid

lurch, 2020-09-19:
@Gadgetoid I've just noticed that the image above (and also used on the live site) has the HAT pins (physical pins 27 and 28) coloured black, which is potentially misleading as they're not ground pins.

Also, the "big diagram" on https://pinout.xyz/ has now changed the colouring of the PCM pins, but both of the "smaller diagrams" are still showing the SPI colouring.

lurch, 2020-09-19:
Also, it's just occurred to that with the symmetrical nature of the Pi Zero, and given that most Pi Zeros come without soldered header pins, it _may_ not be obvious that the diagram for the Zero is from the "top" view rather than the "bottom" view? :shrug: (although perhaps I'm just picking hairs where there aren't any - the main text on the front page of the site _does_ say "orient your Pi such that you are looking at the top")

Other thoughts: Now that you've lost the "rounded corners" from the big diagram, and the green background extends out equally on both sides of the GPIO header, perhaps the green-background is no longer appropriate?

Gadgetoid, 2020-09-20:
Either didn't notice the HAT EEPROM pins were coloured incorrectly, or did it deliberately and forgot why- ha! Good spot. Will have to fix that, along with the new colouring on both graphics.

You're right about the green background being somewhat redundant now. I'm inclined to keep it becausse it's easier than changing it- *but* I also appreciate that the current pale-yellow on green doesn't represent the best visual contrast and readability.

A switch to a high-contrast white background would look something like this and... I don't totally hate it:

[image]

lurch, 2020-09-21:
I wonder if it's also worth shrinking some of the horizontal space between the left-hand labels and the header-graphic? :shrugging:
Might also be worth having "some" kind of coloured background there, to step it blending into the rest of the page? (or maybe that's just because I've only ever seen that part of the page _with_ a background colour)

Gadgetoid, 2020-09-21:
I posted fuller screencaps on https://github.com/Gadgetoid/Pinout.xyz/issues/350#issuecomment-695834925 which show it in context better. It was pretty jarring for me at first, but it's growing on me.

Like you say- when you get used to seeing it one way... it gets difficult to picture it another. The lack of background colour on the pinout itself should - in theory - make it the highest contrast part of the site.

One thing's for sure... making changes is tricky!