pinout-xyz/Pinout.xyz #321: JTAG cannot not use 11 pins at the same time

McTrk, 2019-06-26:
The pinout report for JTAG incorrectly states "Uses 11 GPIO pins".
The correct number is 6 (for Alt 4) or 5 (for Alt 5).
See attached printout for detail.
[JTAG at Raspberry Pi GPIO Pinout.pdf](https://github.com/Gadgetoid/Pinout.xyz/files/3330885/JTAG.at.Raspberry.Pi.GPIO.Pinout.pdf)

lurch, 2019-06-27:
I think that's just an unfortunate side-effect of the way that pinout.xyz "counts" pins.
E.g. https://pinout.xyz/pinout/gpclk says uses 3 pins (they can be used individually), https://pinout.xyz/pinout/ground# says uses 8 pins (they can be used individually), https://pinout.xyz/pinout/i2c# says uses 4 pins (they need to be used 2 at a time), and https://pinout.xyz/pinout/spi# says uses 11 pins (needs to use between 4 and 6 pins).
And in the inverse direction, https://pinout.xyz/pinout/uart# says uses 2 pins, but CTS and RTS are also available on https://pinout.xyz/pinout/pin36_gpio16# and https://pinout.xyz/pinout/pin11_gpio17# if you want to use UART hardware flow-control.

Gadgetoid, 2019-06-28:
I wonder if there's a language change we can make here to cover both cases. Or if this count is even needed at all?

Gadgetoid, 2020-08-20:
I've managed to remove the pin counts from the power and ground pages with a little tweak, this could potentially also apply to JTAG to get rid of this mesage.