pinout-xyz/Pinout.xyz #322: Raspberry Pi 4 Pinout

ccoenen, 2019-06-27:
Here is a reference that is said to be the RPi4's Pin mapping including alternate uses:

[image]

(This was [found by a friend of mine who posted it on twitter](https://twitter.com/timonsku))

lurch, 2019-06-27:
https://github.com/RPi-Distro/raspi-gpio/commit/80fa7d04eafb3ea34fc6f2d32de5f1873b5fb369 is an "official" source of the Pi4 pinout information (which uses BCM-ordering rather than the physical-ordering used above).

shimaore, 2019-07-20:
I suspect the official source is the [datasheet](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0_preliminary.pdf) (page 10) on the Pi website. :)

I would have a go at editing [src/en/template/pinout.yaml](https://github.com/Gadgetoid/Pinout.xyz/blob/master/src/en/template/pinout.yaml) but I'm not sure how the site is supposed to deal with competing Alt values — for example GPIO 4 (pin 7) Alt 4 is currently "AVEIN VID0" but is "TXD3" on the Pi4.

Maybe some new code is needed to handle the Pi4 properly on the site?

lurch, 2019-07-24:
https://www.tomshardware.com/reviews/raspberry-pi-gpio-pinout,6122.html shows one possible way of displaying the multiple Pi4-only functions :man_shrugging:

ccoenen, 2019-07-24:
I find this rather hard to read, to be quite frank.

patrickelectric, 2019-10-09:
hey beautiful people, is there any update related to this issue ?

Gadgetoid, 2020-08-20:
This has been languishing for a while and clearly needs some kind of fix.

Perhaps we need to separate individual board alt functions in the json and display them on the website somewhat like this (very rough example):

[image]

lurch, 2020-08-30:
That table might be easier to read if it _only_ showed where the Pi4 was different, rather than duplicating the majority of the entries which are the same? :man_shrugging:
Also, you'll obviously need to use a different label than "Pi 3" because people will wonder what functions the Pi 2 or Pi Zero uses :wink:

seamusdemora, 2021-06-18:
@lurch :
> https://www.tomshardware.com/reviews/raspberry-pi-gpio-pinout,6122.html shows one _possible_ way of displaying the multiple Pi4-only functions

I also thought this was reasonable, but some things struck me as possibly incorrect: 
  * SDA3 appears on pins #3 and #7 
  * SCL4 appears on pins #21 and #26
  * etc   

But as a format, it works OK.

lurch, 2021-06-18:
> I also thought this was reasonable, but some things struck me as possibly incorrect:
[snip]

That's actually correct, see e.g. https://elinux.org/RPi_BCM2711_GPIOs  (although note that eLinux is a non-authoritative reference :wink: )

seamusdemora, 2021-06-19:
Yes! I realized that a few hours after posting this comment, and failed to follow up - all part of the GPIO musical chairs!

Gadgetoid, 2026-08-19:
Added tabbed alt mode tables for all the things. Though I'm not sure this helps the headline pinout pages much.

closed 2026-08-19 by Gadgetoid