pinout-xyz/Pinout.xyz #331: Update Pinout.xyz codebase to Python3

vwillcox, 2019-09-10:
Since Python3 is now the standard, it would be good to try and get this code updated.

I will work on this and post updates, found a few gotchas

for example to use

`reload(sys)`

you now need to  import a new lib

`from importlib import reload`

unicode has also changed

Gadgetoid, 2019-10-05:
I ran into problems with this recently (well a few months ago... time seems to be a blur right now) when I was setting up Pinout tooling on a new computer. You're right- it's long overdue an update and probably some other quality-of-life fixes.

Gadgetoid, 2019-10-10:
I believe I've managed to get it running reliably on Python- I used a diff tool to compare the Python 2 and 3 output, and then afterwards to check my subsequent clean up/fixes.

lurch, 2019-10-10:
I guess #294 will need to be updated for Python3 too, when the switch happens :wink:

closed 2026-08-18 by Gadgetoid