pinout-xyz/Pinout.xyz #337: The clock divisor formula on datasheet is wrong?

carlosbfeitozafilho, 2019-10-03:
Your site (https://pinout.xyz/pinout/gpclk) states that there are an error at DataSheet page 105 and that the correct divisor is 4096 and not 1024. The site do not show any source for this information and I would like to know if this is true. If so, please, show me some sources

lurch, 2019-10-03:
There's more info at https://elinux.org/BCM2835_datasheet_errata#p105_table but I think you'll need to decide for yourself how reliable the (community-contributed) info is?

carlosbfeitozafilho, 2019-10-04:
> There's more info at https://elinux.org/BCM2835_datasheet_errata#p105_table but I think you'll need to decide for yourself how reliable the (community-contributed) info is?

Hi
Hi

Well, I only need a source for that information because I'm making my own library and this subject (clock divisor) is a little bit obscure. Some points at https://elinux.org/BCM2835_datasheet_errata#p105_table are logic and so, thrustfull, like the exchanged columns "min output frequency" and "max output frequency", but the main point (4096 or 1024) yet is confuse. The information source for that (https://github.com/arisena-com/rpi_src.git) states that the correct value is 4096 but the source code used like example uses 1024! Well the only source for this information affirms that the correct value is one, but continues to use the original, and socalled wrong, value.

lurch, 2019-10-04:
:man_shrugging: 

ping @arisena-com in case he's able to offer any further insight.
Perhaps you'll need to perform some experimentation yourself, using e.g. an oscilloscope to check the output?

carlosbfeitozafilho, 2019-10-04:
> 🤷‍♂
> 
> ping @arisena-com in case he's able to offer any further insight.
> Perhaps you'll need to perform some experimentation yourself, using e.g. an oscilloscope to check the output?

Would be great, but I do not have one :(

Well, I'll try to contact the author of https://github.com/arisena-com/rpi_src.git. Maybe he can say why the value is wrong but he continue using it.

Thanks for trying to help

carlosbfeitozafilho, 2019-10-07:
Unfortunately I could not contact author by e-mail. Some kind of mail problem. I'll try to send it again. If someone have more info or another way to contact the author, please, let me know

carlosbfeitozafilho, 2019-10-09:
> Unfortunately I could not contact author by e-mail. Some kind of mail problem. I'll try to send it again. If someone have more info or another way to contact the author, please, let me know

Definitively the author is inaccessible, so I guess the information about the clock divisor is not as precise to be considered true, despite the fact that it has been written at https://elinux.org/BCM2835_datasheet_errata. If someone can explain *clearly* why the divisor is 4096 instead 1024, the information can be stated in https://pinout.xyz/pinout/gpclk. Until that be made, no one can assume that is true that the correct value is 4096.

I recomend to remove this information from https://pinout.xyz/pinout/gpclk or edit it to make it mere suppositions to be accepted "at your own risk"

lurch, 2019-10-09:
https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/clk/bcm/clk-bcm2835.c says `# define CM_DIV_FRAC_BITS	12` and 2**12 is 4096, but I don't know if that's relevant info or if I'm just clutching at straws?? :laughing:

carlosbfeitozafilho, 2019-10-09:
> https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/clk/bcm/clk-bcm2835.c says `# define CM_DIV_FRAC_BITS 12` and 2**12 is 4096, but I don't know if that's relevant info or if I'm just clutching at straws??

Well, as I understand, the DIVI and DIVF are two 12 bits fields from the clock divisor registers. There are 3 general purpose clocks plus 2 "special" clocks (PWM clock and PCM clock), totalizing 5 registers to configure the divisor, one for each available clock. As 12 bits registers they can store up 4095, not 4096.

The CM_DIV_FRAC_BITS constant is only used to do binary math, so, unfortunately you are only "clutching at straws" ;)

I'm concluding that no one ever work with fractional clock divisors and this seems true because the most of examples never use fractional divisors, only integer ones :/ It's a little bit sad that conclusion, because I'm basing my library on WiringPi, BCM2835, RPI_HAL and PiGPIO and none of them treats this subject clearly

lurch, 2019-10-10:
Ping @joan2937 as you mentioned PiGPIO and they seem to have a fairly solid understanding of the low-level hardware on the RPi.

joan2937, 2019-10-10:
Page 108 of [BCM2835 ARM Peripherals](http://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf) shows 12 bit fields for the integer and fractional divisors.

Experiment has shown that valid values are 0 to 4095 and these are the values used by pigpio.  A divisor of 0 results in an actual divisor of 4096.

I don't remember how I became aware, probably the eLinux errata linked above.

carlosbfeitozafilho, 2019-10-14:
> Page 108 of [BCM2835 ARM Peripherals](http://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf) shows 12 bit fields for the integer and fractional divisors. [...]

Hi @joan2937 and thank you to join this conversation. We are talking about the average clock formula on page 105 of the datasheet. Someone affirms that the value shown at formula would be 4096, instead 1024 like the datasheet says. Please read the entire conversation here to understand more

joan2937, 2019-10-15:
@carlosbfeitozafilho No one can force you to believe what you are told.  If you don't believe the divisor is 4096 then use any value you want.  Why don't you just try different values until you find one you are happy with?

carlosbfeitozafilho, 2019-10-16:
**_> @carlosbfeitozafilho No one can force you to believe what you are told. If you don't believe the divisor is 4096 then use any value you want._**

Well, I believe in facts and well explained theories. The theory about the 4096 was explained by @arisena-com and I tried to contact him, because despite his affirmation, his source-code (used to explain, I guess) continues to use 1024 on the refered formula. So, why I (or anyone else) could assume this as "correct"?

If you have a clear explanation about this I would be thankful and I will imediatelly stop bother you all with my "truth hunting"

_**> Why don't you just try different values until you find one you are happy with?**_

Because this is not the matter... I can use any integer divisor I want (2 to 4095), but I would like to make my lib as most generic as I can. As I said before I'm calculating the PWM Range using the folowing formula:

`RangeValue = PWMClockSourceHertz / PWMClockDivisor / RepeatCycle;`

In this formula PWMClockSourceHertz / PWMClockDivisor gives the output frequency and I guess the this frequency must be the average value based on DIVI and DIVF and the page 105 says that the average clock divisor is obtained by the formula:

`DIVI + DIVF/1024`

Now, PWMClockDivisor can be DIVI + DIVF/1024 following the datasheet or DIVI + DIVF/4096 following @arisena-com's thoughts. In my tests (and I admit that they are not good ones), exchanging 1024 with 4096 produces no aparent differences on PWM configuration. Because this I'm here. The source for the 4096 information is uncertain. I do not know Mr. @arisena-com, maybe he is a great authority on RPI development. If so, here comes my most sincere apology.

For now, I will follow the datasheet and assume It is correct using 1024 and like a final advice, i guess the pinout.xyz is very important to reproduce not checked informations. Please thing about it.

closed 2019-10-16 by carlosbfeitozafilho