pinout-xyz/Pinout.xyz #342: Why the first 3.3v pin (physical pin 1) representation is the only one with square shape?

carlosbfeitozafilho, 2019-10-09:
My OCD is driving me crazy because of this...

Well, just joking, but there are a reason why this is the only one pin with square shape representation?

Gadgetoid, 2019-10-09:
It's the pin 1 designator. I appreciate there are also numbers, but using a square for pin 1 is... somewhat... canonical for header pinouts.

lurch, 2019-10-10:
If you look at the underside of the GPIO header on an actual RPi, you'll see that Pin1 has a square solder-pad too, whilst all the others are round.
See e.g. the second image on this page https://www.raspberrypi.org/products/raspberry-pi-zero/

lurch, 2019-10-10:
See also #267

carlosbfeitozafilho, 2019-10-16:
Thank you guys!
Now I understand :)

lurch, 2019-10-16:
> Now I understand :)

Can this issue be closed then? :)

closed 2019-10-27 by carlosbfeitozafilho