pinout-xyz/Pinout.xyz #344: No mention of I2S

lurch, 2019-10-10:
Hopefully an easy fix - perhaps https://pinout.xyz/pinout/pcm should also mention the acronym I2S, since I suspect "the I2S bus" is a more common name than "the PCM bus" ?

Gadgetoid, 2019-10-10:
I think you're right. I wonder if I can make the text link read "PCM/I2S" without hacking the codebase too much.

MSBGit, 2022-04-16:
I have started working with the Adafruit i2s MEMS microphone and their [RPI page on this device](https://learn.adafruit.com/adafruit-i2s-mems-microphone-breakout/raspberry-pi-wiring-test) has a good description of how to connect the microphone to the i2s bus. Having never worked with i2s before it was a bit confusing at first to not see _i2s_ mentioned in the pinout page but to see _PWM_ mentioned. It would be a great enhancement to add an _i2s_ tab that would highlight the signals used for i2s devices.

closed 2026-08-19 by Gadgetoid