pinout-xyz/Pinout.xyz #346: More explicitly state 40-pin is superset of original 26-pin

RalphCorderoy, 2019-10-12:
Wanting to modify an existing old-Pi installation, I find a 26-pins. My recollection is it's a strict subset of the current 40-pins, but it would be nice if the site made that explicit near the 'Rev 1 Pi' on the main page. (I couldn't find those alternative BCM numbers mentioned there.)

dmatherpi, 2020-03-11:
I agree, this would be good to clarify on the page.

Jalle19, 2021-02-18:
Being a GPIO noob, I had to look for other resources to confirm which pins were which on an old 26-pin Raspberry Pi I did some experimentation with. Some form of toggle somewhere to show the 26-pin layout would be nice, or any method really that illustrates the difference.

closed 2026-08-19 by Gadgetoid