pinout-xyz/Pinout.xyz #352: Wrong pinouts for SenseHat

docnjt, 2019-12-21:
Your diagram for the pin outs on the Pi Sensehat needs correcting - those in particular 27,28. 

the correct pins should be:

1, 2 - 3v and 5v power
3, 5 - BCM 2 and 3 - I2C
16, 18 - BCM 23 and 24 - Joystick
**27, 28 - BCM 0 and 1 - EEPROM**
Ground

I was wondering why my sense hat wasn't working via jumper leads!!

lurch, 2019-12-22:
Duplicate of #223 ?

docnjt, 2019-12-26:
yes but the pinout diagram has not been corrected at:?? 

https://pinout.xyz/pinout/sense_hat

bsimmo, 2020-03-29:
22/24 are not wrong, they are on the sensehat. Just not really used.
I think the argument in the previous issue was that the hat will work without the eeprom connected (if you add the overlay yourself).
Much like the other devices with eeprom will.
Personally I would like to see them all marked, but in the description what they do/cavats etc.
Unfortunately it was a long time ago since I last looked at a sensehat to remember properly.

arnitdo, 2020-08-27:
I recently tried connecting the Sense Ha to the Raspi using jumper wires.
I followed the official schematic given by the Raspberry Pi Foundation and got it working . [rpi-sense](https://www.raspberrypi.org/documentation/hardware/sense-hat/images/Sense-HAT-V1_0.pdf)
According to the drawing, it also uses GPIO 10,9,11 pins (MOSI, MISO, SCK).
apart from GPIO 2,3,23,24,25,8, as well as 5v, 3v3 and GND pins
The three pins (GPIO 10,9,11) are not shown / highlitghted on the website.
It seems they are necessary for the SenseHat to work
Please look into it.

Gadgetoid, 2020-08-27:
Any source I can find showing the Sense HAT in use suggests that the SPI pins (MOSI, MISO, SCK) are not used unless the onboard firmware is being reflashed. I don't understand how they can be necessary for normal functionality, but will add it to the TODO pile to look into. I don't have a Sense HAT for testing, though, so it might be handy for some other users to weigh in.

lurch, 2020-08-30:
I've just tried looking into this... looking at the schematics there's actually two "firmware chips" on the Sense HAT. One being the internal flash on the ATTINY88 microcontroller (connected to the Pi via SPI) and the other being the traditional HAT EEPROM (connected to the Pi via I2C).
Looking at https://www.raspberrypi.org/documentation/hardware/sense-hat/README.md there's instructions for reflashing the EEPROM (using `eepflash.sh`), but the instructions for reflashing the ATTINY88 just say `Updating the AVR firmware ...` !  So maybe the original intention was that the ATTINY88 would be user-reprogrammable, but then a decision was made that this was simply unnecessary / would add too much confusion ? :shrug:
(Ah! After much googling I found https://github.com/raspberrypi/rpi-sense which clearly doesn't have much activity going on)

So while it's probably worth leaving the "In addition, the Atmel chip can be reprogrammed via the SPI interface" text on the pinout.xyz page, it probably makes sense to remove GPIO25 and GPIO8 from the list of pins "required" by the Sense HAT?
(i.e. IMHO you should either document GPIOs 10, 9, 11, 25 and 8 as all being used, or mark **none** of those pins as being used; with the latter being the more sensible option).

Also, in the "Details" section you should change `0x46: LED2472G` to `0x46: ATTINY88`, and it's probably worth labelling https://github.com/RPi-Distro/python-sense-hat as "Python GitHub Repository" and adding https://github.com/raspberrypi/rpi-sense as a link to "Firmware GitHub Repository" ? And might be worth adding a link to https://www.raspberrypi.org/documentation/hardware/sense-hat/README.md too?

I've just done a test with my SenseHAT, and with `dtoverlay=rpi-sense` added to the bottom of `/boot/config.txt` I can confirm that it seems to work fine with only pins 1, 2, 3, 5, 6, 16 & 18 connected via jumper-wires.

Ah, and I've just noticed that https://github.com/RPi-Distro/python-sense-hat now auto-redirects to https://github.com/astro-pi/python-sense-hat

closed 2026-08-19 by Gadgetoid