pinout-xyz/Pinout.xyz #354: Set CSS background and color to default

Zocker1999NET, 2020-01-28:
The page does not configure the `background` and `color` properties to default first, it instead expects that the user has not modified his default values for background and text color. Cause I changed these values so websites without an unique proper style will be adapted to the style of my desktop, see:
[image]
However the style of this page is too customized to can be adaptive to the user configuration, so I would propose to define these properties.

I would propose this change by myself, however I do not know to which file I should add these properties cause there are three style sheet files in the `resources` directory: `pinout.css`, `pinout.scss` and `pinout.scss.css`.

Gadgetoid, 2020-08-18:
Hopefully this is now fixed! Or at least mostly fixed.

closed 2020-08-18 by Gadgetoid

Zocker1999NET, 2020-08-18:
Is fixed, thanks @Gadgetoid