pinout-xyz/Pinout.xyz #355: Pi-Plates

pi-plates, 2020-02-11:
Can you add the Pi-Plates family to your https://pinout.xyz/boards page?

https://pi-plates.com

lurch, 2020-02-11:
Totally off-topic.... but I just looked at https://pi-plates.com and it seems **very** odd that you can't see the prices of any of the products until _after_ you've added them to your shopping cart?? :confused: 

> Can you add the Pi-Plates family

Given how busy @Gadgetoid is, and how many other [open requests](https://github.com/Gadgetoid/Pinout.xyz/issues?q=is%3Aopen+is%3Aissue) there are, I suspect you'll _probably_ get a faster response if you [submit a PR](https://github.com/Gadgetoid/Pinout.xyz#contributing) yourself? :man_shrugging:

pi-plates, 2020-02-11:
You can't see the prices on this page?

https://pi-plates.com/pi-plates-catalog/

Jerry Wasinger
Founder, Pi-Plates

lurch, 2020-02-11:
Hmmm, I can see prices on that page, but not on e.g. https://pi-plates.com/product/motorkit1/ or https://pi-plates.com/product/motorplate/ :confused:

pi-plates, 2020-02-11:
I'm curious: how did you reach those pages? They are basically old
placeholders required by my ecommerce software. The MOTORplate product page
is here:  https://pi-plates.com/motorr1/

Jerry Wasinger
Founder, Pi-Plates

lurch, 2020-02-11:
https://pi-plates.com/ has a "The MOTOR KIT:" link - I clicked that to get to https://pi-plates.com/product/motorkit1/ (which doesn't display a price) and from there I clicked on the "MOTORplate" link to get to https://pi-plates.com/product/motorplate/ (which also doesn't display a price).

Gadgetoid, 2020-02-22:
I'm afraid @lurch is correct- as you can tell by the somewhat neglected backlog of Pinout.xyz I've been whole new levels of busy lately. A pull-request would be most appreciated- since you have a better knowledge of your products and their pinout than I could hope to gain quickly.

pi-plates, 2020-02-22:
Thanks, Philip!

Jerry Wasinger
Founder, Pi-Plates

closed 2026-08-18 by Gadgetoid