pinout-xyz/Pinout.xyz #359: RaspiAudio Mic+ also needs GPIO 20 / pin 38 for the microphone

HorstBaerbel, 2020-03-27:
See [here](https://www.raspiaudio.com/raspiaudio-aiy). This is missing in [https://pinout.xyz/pinout/raspiaudio_audio_speakers_mic#](https://pinout.xyz/pinout/raspiaudio_audio_speakers_mic#).
Also it will use GPIO 16 / pin 36 if you use the Google AIY driver.

lurch, 2020-03-27:
@roscop13 added this board initially (in #270 ), so giving him a quick ping in case he has any comments :slightly_smiling_face:

Gadgetoid, 2020-06-24:
Amazing, the linked resource uses a screencap with amendments instead of raising a PR 😆 

I think I should probably fix this!

Gadgetoid, 2020-06-24:
Should be fixed, pending flushing through the CloudFlare cache- https://pinout.xyz/pinout/raspiaudio_audio_speakers_mic?v=1

closed 2020-06-24 by HorstBaerbel