pinout-xyz/Pinout.xyz #364: please rename SPI signal names

ecc1, 2020-07-08:
https://www.oshwa.org/a-resolution-to-redefine-spi-signal-names

Gadgetoid, 2020-07-09:
While I don't disagree with this change, you could perhaps have made the request with slightly more tact.

In future I implore you to be more conscientious in your approach to making these requests of project owners. You do not endear people to a cause by taking just 5 seconds to paste a link with no achknowledgement of the issue, the project, or the wider scope of the terminology change.

ecc1, 2020-07-09:
Sorry, the link I included discusses the background and rationale far more
eloquently than I could have. I apologize if not providing an excerpt of
that in-line came across as tactless.

Gadgetoid, 2020-07-09:
No worries- for better or worse it got the ball rolling on making, discussing and refining these changes and Pinout has the sort of exposure where - even if we might not necessarily completely agree with the reasoning behind the changes - we stand to make a positive impact on others understanding of how the old terminology maps to the new.

closed 2020-08-19 by Gadgetoid