pinout-xyz/Pinout.xyz #366: New home for GPIO library list?

cinderblock, 2020-07-12:
Hello,

I love your site and use it all the time.

One problem I've run into in working with GPIO on Raspberry Pi is the shear number of libraries that are available for doing GPIO. Curiously, if you follow the tree of dependencies, they all really only use one of the ~2 low level APIs (DMA or GPIO registers). Some even depend on one library that itself talks to another library.

I started to make a list of libraries that I'd found and taken the time to look at their source code enough to figure out how they worked internally. Please take a look if interested https://docs.google.com/spreadsheets/d/1sFCJuPZ9k5GN0A6j7gHRckvbwIdnwzPH1sTCuPNFNRQ/edit?usp=sharing

I'm writing here now because I thought maybe you might like to try incorporating this information into your site. If not, that's frankly expected. But in the off chance you're interested, feel free to use the information that I've collected. Maybe someone else here finds it useful too!

Cheers!

Gadgetoid, 2020-07-14:
I don't know why I expected that list to be shorter- at least five of those are my fault 😆 

This is an interesting overview- I wonder how it could be presented to be useful to users. Possibly dilineated by language for starters.

Anyway glad you love the site, and thanks for compiling and sharing this list. I'll have a think about if/where it fits in, though I'm not so sure I'll find time to do anything with it in the short run- everything is still absolute chaos!

lurch, 2020-09-19:
Nice collection of information, I can appreciate that it must have taken quite a while to pull together! :+1: 

I _suspect_ that many of those projects will have been released in the early days of the Pi, and will now be defunct / unmaintained / abandoned. Perhaps "supports Pi4" would be a good way to differentiate "active" projects from "inactive" projects, given that the Pi4 was now released over a year ago?

BTW gpiozero should definitely be listed as "popular", it uses the "Broadcom" numbering scheme, and can either use RPi.GPIO, pigpio or RPIO backends, or can also use it's own "Memory"-based native backend https://gpiozero.readthedocs.io/en/stable/api_pins.html . It also supports SPI https://gpiozero.readthedocs.io/en/stable/api_spi.html . Version history: https://gpiozero.readthedocs.io/en/stable/changelog.html

BTW there's also https://github.com/quick2wire/quick2wire-python-api and https://github.com/rakeshpai/pi-gpio and https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/circuitpython-raspi if you want to add even more to your list :wink:

cinderblock, 2020-09-20:
@lurch thanks!

The list is publicly editable. Please add/change as you see fit!

lurch, 2020-09-20:
That's okay, I'll leave it up to you to update :wink:

Gadgetoid, 2026-08-19:
Thank you. However, the ceaseless churn of GPIO libraries is not a burden I want to take up here- I've had enough of them to last me a lifetime 😆

closed 2026-08-19 by Gadgetoid

lurch, 2026-08-19:
See also the "A history of GPIO usage on Raspberry Pi devices, and current best practices" document at https://pip.raspberrypi.com/categories/685-app-notes-guides-whitepapers