pinout-xyz/Pinout.xyz #368: GPIO18 labeled a s(duplicate) PWM0

gogglespisano, 2020-08-07:
GPIO18 should be PCM CLK, not PWM0.  GPIO12 is already PWM0.

lurch, 2020-08-08:
Each pin has multiple functions e.g. PWM0 is ALT5 on GPIO18 https://pinout.xyz/pinout/pin12_gpio18

I guess which function gets labelled onto which pin on the diagram on the left-hand side of https://pinout.xyz/pinout/pin12_gpio18 is somewhat arbitrary?
https://pinout.xyz/pinout/pcm shows the PCM-related pins.

gogglespisano, 2020-08-08:
How about have the non-specific version of the pinout show the primary function for every pin rather than some having nothing or an alternate, and at the very least, not have any pins labeled exactly the same?

lurch, 2020-08-08:
Just to be pedantic, the "primary function" of each pin is just software-controlled GPIO (i.e. no label at all) :wink:

Not much more I can add to this conversation - it's up to @Gadgetoid to decide.

gogglespisano, 2020-08-08:
By primary, I mean the Raspberry Pi labels.

https://www.raspberrypi.org/documentation/usage/gpio/README.md

lurch, 2020-08-08:
That's interesting... the Raspberry Pi documentation page you link to labels GPIOs 19, 20 & 21 as PCM, whereas Pinout.xyz labels them as SPI1. (IMHO each function is equally valid; it's probably just personal preference which function you choose to label as more important)

Gadgetoid, 2020-08-08:
I can't recall *why* I picked that particular label, but it's probably strongly linked to Pimoroni breaking out that pin and labelling it "PWM" on HAT expansion headers and/or using it to drive ws281x pixels.

In lieu of not being able to provide a strong opinion about why we should stay as we are, I can only defer to @gogglespisano's observation that we should probably match the official docs.

Of course I'm not sure when I'll get a moment to implement this!

gogglespisano, 2020-08-08:
After searching around trying to decide what the "primary" labels should be I ended up using the ALT0 labels on page 10 of https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0_preliminary.pdf.

@Gadgetoid Thank you for making this site.  Quickly seeing the functional variants was just what I needed to choose the GPIO assignments for a project.

Gadgetoid, 2020-08-08:
That seems at least consistent! I'll flag this up as an Enhancement and hopefully find the time to do it sooner rather than later (tackling CircuitPython at the moment is draining my mental resources).

And thank you- though I don't deserve the thanks alone! I've had a lot of help and support over the years to get Pinout where it is today. In fact... you're helping and supporting right now 😆

lurch, 2020-08-08:
Be careful that some of the ALT functions on the [BCM2711](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0.pdf) are slightly different to the ALT functions on the [BCM283x](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf) - see also #322 .

closed 2026-08-18 by Gadgetoid