pinout-xyz/Pinout.xyz #369: Adding new items to the phatstack list - specifically GFX hat

qbert911, 2020-08-09:
I want to contribute to the codebase but I'm having a devil of a time finding how to add a hat to the list on the phatstack page.

Please assist.

Gadgetoid, 2020-08-09:
I don't think pHAT Stack was ever completely finished, so it's not baked into the workflow for Pinout.

It basically has *all* the board data stored in the HTML and figures out which boards to display by filtering out HAT, pHAT and Custom types with a "manufacturer" of "Pimoroni".

IIRC `generate-json.py` is the script that works the magic, but the result is currently manually copied and pasted into the pHAT Stack HTML.

I limited to Pimoroni boards since those were the ones I could ensure I had good enoug top-down photos of to make the concept work, but there's no reason the limit should stay that way.

You could probably test locally by generating the json, pasting it in and tweaking the Javascript to display additional boards (if you need other manufacturers) but it's all *very* manual right now.

qbert911, 2020-08-10:
This feels beyond the scope of my comprehension sadly.
Please add new Pimoroni GFX hat (already on pinout) to the stacker.