pinout-xyz/Pinout.xyz #372: SPI Signal

TerryJColes, 2020-08-17:
Hi,

I raised a query over at the RPI Forums because I found the labelling of the SPI Bus pinouts confusing [https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=282970&p=1713753#p1713753](url).  In essence, I couldn't understand why the RPI Hardware Spec said that there was only one SPI Bus, whereas the Pinouts shown on the SPI page appeared to show two busses (SPI0 and SPI1).

I was informed in the responses that the second SPI Bus is not active.  If this is the case could the additional labels be removed or annotated to indicate that the functionality isn't available?

Gadgetoid, 2020-08-17:
The `spi1` bus can be enabled by adding `dtoverlay=spi1-3cs` to `/boot/config.txt`. Since neither SPI0 or SPI1 are enabled *by default* I don't see any particular reason why we can't keep the labels- Dougie saying SPI1 is "normally inactive" is disinginuous when SPI0 is *also* "normally inactive." Yeah I'm splitting hairs 😆 

*However* we should probably document at least one way to enable that bus- ie the dtoverlay mentioned above.

From prior experience with SPI1 it has some functional limitations, I'm not sure if these apply specifically to generic SPI use but they ruined my efforts to drive WS281x pixels. It might be worth touching upon those, too.

TerryJColes, 2020-08-17:
Yes.  I've just read Dougie's response about the overlays.  Also splitting hairs; SPI0 can be enabled very easily by rookie users like me; simply by using raspi-config, but the other channels need someone with a bit more RPi smarts.

Having said that, there is a case for showing them, so can I modify my request to reflect the above remarks and provide the info that the full functionality available can be found in /boot/overlays/README?

Also, under the Details heading there is a statement that the function uses 11 GPIO pins.  This is only true when the relevant overlay has been enabled.

lurch, 2020-08-18:
Yup, the SPI0 and SPI1 are in different [hardware blocks](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf) and so have different capabilities (with SPI0 being the more fully-featured interface, IIRC). That document also refers to SPI2 (which is of a similar type to SPI1), but that's only on pins which [aren't available](https://elinux.org/RPi_BCM2835_GPIOs) on the Pi's 40-pin header, so outside the remit of pinout.xyz :slightly_smiling_face: 

But note also that there are multiple 'full-featured' SPI interfaces on the [Pi4](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0.pdf) : SPI0, SPI3, SPI4, SPI5, SPI6.

TerryJColes, 2020-08-18:
> (with SPI0 being the more fully-featured interface, IIRC).
I may have misunderstood, but my interpretation is that SPO0 only has two Chip Select (CE) ins, whereas SPO1 has three.

Or is that not what you meant?

lurch, 2020-08-18:
I've just had a quick look at https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/ and unfortunately much of the info on that page is **very** out of date. (there's one part of the page that says "The following information was valid 2014-07-05."!)
It looks like most of that page was written after the release of the original 26-pin Model B, but before the release of the 40-pin Model B+, and at that time the sentence "The BCM2835 on the Raspberry Pi has 3 SPI Controllers. Only the SPI0 controller is available on the header." was accurate; but on 40-pin Raspberry Pis both SPI0 and SPI1 are available on the header.
Feel free to submit an issue or PR at https://github.com/raspberrypi/documentation to correct this.

> > (with SPI0 being the more fully-featured interface, IIRC).

> I may have misunderstood, but my interpretation is that SPO0 only has two Chip Select (CE) pins, whereas SPO1 has three.
Or is that not what you meant?

The differences between SPI0 and SPI1 go much deeper than just the number of chip-select pins. I dunno much about SPI myself, so for full details you'll need to read chapter 10 (SPI0) and chapter 2 (SPI1) of the [datasheet](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf).

TerryJColes, 2020-08-18:
> Feel free to submit an issue or PR at https://github.com/raspberrypi/documentation to correct this.

Done.

DougieLawson, 2020-08-25:
> The `spi1` bus can be enabled by adding `dtoverlay=spi1-3cs` to `/boot/config.txt`. Since neither SPI0 or SPI1 are enabled _by default_ I don't see any particular reason why we can't keep the [...]

If I pile into sudo raspi-config and select SPI that way it runs
`  set_config_var dtparam=spi $SETTING $CONFIG &&` 
and that results in 
```
dtparam=spi=on
```
being tagged on the bottom of /boot/config.txt (nuffink about SPI1 in there). So I'll stick with saying that SPI1 is "normally inactive" because it needs more that a beginner's knowledge to activate it.

Gadgetoid, 2020-08-25:
@DougieLawson oh heck, you caught me 😆 

But you're right- I think Pinout often ends up in a tug-of-war between beginner-friendly documentation and containing slightly more advanced stuff. It should err on the former.

As part of my up-too-late-last-night sprint I did some tweaks vaguely relating to this thread and removed SPI1 in favour of PWM/I2S. I think the number of boards that actually *use* SPI1 is somewhere in the region of zero, whereas the number that use I2S is... many. And SPI is still right there on the interface tab.

Thanks for weighing in! Stuff gets out of hand if people don't keep me grounded.

DougieLawson, 2020-08-25:
@Gadgetoid is there a beta test of the latest changes or are you just pushing them straight out to the public website? Only https://pinout.xyz/pinout/spi still shows SPI1 but the doc box on the right hand side has no mention of what it is, why it exists or how you activate it. 

While we're here you should consider turning all of the wiringPi stuff from present tense to past tense (like they do for folks who die on Wikipedia).

Gadgetoid, 2020-08-25:
@DougieLawson I've raised myself an issue + TODO list to address the details pages- there's a *lot* of old cruft to overhaul. See: #375

Changes so far *only* concern the pinout diagram's default state and text labels.

TerryJColes, 2020-09-17:
I can now report that the SPI Page at [](https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md) has now been updated to reflect the correct hardware arrangements.

Many thanks to all who contributed to this change.

lurch, 2020-09-19:
Can this issue be closed then, or are there still clarifications that you'd like pinout.xyz to make?

TerryJColes, 2020-09-19:
Yes.  I think that it probably can, since the updated page includes a link to the updated SPI Page.

closed 2026-08-18 by Gadgetoid